evcc-io/evcc #23453: Homeassistant vehicle: Wrong documentation of website?

greimel, 2025-09-07:
### Describe the bug

According to the website, you can list all properties like

```
vehicles:
  - name: my_car
    type: template
    template: homeassistant
    title: # Titel, Wird in der Benutzeroberfläche angezeigt (optional)
    capacity: 50 # Akkukapazität (kWh), optional
    uri: http://192.168.1.10:8123 # Home Assistant URI, HTTP(S) Adresse
    token: eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9... # Home Assistant Long-Lived Access Token
    soc: sensor.vehicle_soc # Ladezustand [%]
    range: sensor.vehicle_range # Restreichweite [km], optional
    status: sensor.vehicle_charging # Ladestatus, optional
    limitSoc: number.vehicle_target_state_of_charge # Ziel-Ladezustand [%], optional
    odometer: sensor.vehicle_odometer # Kilometerstand [km], optional
    climater: binary_sensor.vehicle_climater # Klimatisierung aktiv, optional
    finishTime: sensor.vehicle_finish_time # Ladeende (ISO8601 oder Unix), optional
    start_charging: script.vehicle_start_charge # Service zum Laden starten, optional
    stop_charging: script.vehicle_stop_charge # Service zum Laden stoppen, optional
    wakeup: script.vehicle_wakeup # Service zum Aufwecken, optional
```

By contrast, the [discussion linked in the blog](https://github.com/evcc-io/evcc/discussions/22676) separates the sensors and scripts like this.

```
vehicles:
  - name: mercedes_eqs
    title: "Mercedes EQS 580"
    type: homeassistant
    uri: http://homeassistant.local:8123
    token: "eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9..."  # Langzeit-Token
    
    sensors:
      soc: sensor.eqs_580_state_of_charge           # PFLICHT: SoC in %
      range: sensor.eqs_580_range_electric          # OPTIONAL: Reichweite in km
      status: sensor.eqs_580_charging_status        # OPTIONAL: Ladestatus
      limitSoc: number.eqs_580_max_soc              # OPTIONAL: Ladelimit in %
      odometer: sensor.eqs_580_odometer             # OPTIONAL: Kilometerstand in km
      climater: climate.eqs_580_thermostat          # OPTIONAL: Klimaanlage (falls verfügbar)
      finishTime: sensor.eqs_580_charging_end_time  # OPTIONAL: Ladeende-Zeit (falls verfügbar)
    
    services:
      start_charging: script.eqs_start_charging     # OPTIONAL: Script zum Laden starten
      stop_charging: script.eqs_stop_charging       # OPTIONAL: Script zum Laden stoppen  
      wakeup: script.eqs_wakeup                     # OPTIONAL: Script zum Aufwecken
    
    capacity: 107.8  # Batteriekapazität in kWh
```

My test suggests that the second version is correct.

### Steps to reproduce

1. bla
2. bla
3. bla
...

### Configuration details

```yaml
blabla
```

### Log details

```text
blabla
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-09-08:
These are equivalent.

closed 2025-09-08 by andig

andig, 2025-09-08:
It would- as usual- help to share the relevant information. in this case: the actual error message.

greimel, 2025-09-08:
works:

```
vehicles:
  - name: zoe_2017
    title: "Zoe 2017"
    type: homeassistant
    uri: http://homeassistant.local:8123
    token: ""
    sensors:
      soc: sensor.zoe_soc_filtered
      odometer: sensor.wican_odometer
    #limitSoc: 95
    capacity: 34 # 40 * 0.85
```

errors:

```
vehicles:
  - name: zoe_2017
    title: "Zoe 2017"
    type: homeassistant
    uri: http://homeassistant.local:8123
    token: ""
    #sensors:
    soc: sensor.zoe_soc_filtered
    odometer: sensor.wican_odometer
    #limitSoc: 95
    capacity: 34 # 40 * 0.85
```

error message:

[image]

greimel, 2025-09-08:
Probably it is relevant that adding other fields leads to errors as well (note `minCurrent`, `maxCurrent` and `limitSoc`:

```
vehicles:
  - name: zoe_2017
    title: "Zoe 2017"
    type: homeassistant
    uri: http://homeassistant.local:8123
    token: "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJhZmYxMmZmYmU4MDg0NzdlOTk1YjU0NDAxZTdhZGI5OCIsImlhdCI6MTc1NzI3NjI4NiwiZXhwIjoyMDcyNjM2Mjg2fQ.vA-I2ZI1uz4KdAKvtkCb6WL_i-AZypsP7AHADamLFQc"
    sensors:
      soc: sensor.zoe_soc_filtered
      odometer: sensor.wican_odometer
    limitSoc: 95
    capacity: 34 # 40 * 0.85
    #icon: car # Icon in der Benutzeroberfläche (optional)
    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
    minCurrent: 8 # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    maxCurrent: 32 # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
```

error:

[image]

andig, 2025-09-08:
That's simply not what you've shown above:

    type: template
    template: homeassistant

Just use the UI for config and things will work.

greimel, 2025-09-08:
Sorry, my bad. Thanks for clearing that up!