evcc-io/evcc #23455: A charging plan like:"At any time, aim to charge to X percent within the next Y hours"

KonstantinSchubert, 2025-09-08:
**Is your feature request related to a problem? Please describe.**
I want to charge my car with the cheapest power available. I have little solar, but a smart grid tariff. 
No matter what target time I set in my charge plan to, if there is cheap power right after that, it will not be used, because the plan is designed to always meet its goal. But I would be happy to wait sometimes. 
For example, if my target time is 8am, and I plug in the car in the evening, the car will charge overnight even though there is cheap power at noon and I am happy with my min-charge being applied. The same problem can happy with any other target time I choose, depending on how the prices are on that particular day. 

**Describe the solution you'd like**
I would like the option to define my charging plan more dynamically. Something like "always plan the charge such that the car would reach X percentage in Y hours or less".  As the time moves, the charging will be re-planned according to the aim of "in the next 24 hours". 
In other words, if it's two PM now, and Y=24 hours,  the planner will aim to charge with the cheapest power until 2pm tomorrow. At 3pm, it will re-plan to charge with the cheapest power until 3pm tomorrow. That should make much better use of cheap power than a fixed charging goal like 8am. 

**Describe alternatives you've considered**
One option would be to dynamically adjust the "always charge below this price" limit dynamically based on the current electricity market. But that does not play well if there is also power coming from PV. 

I may try to implement this, at least as a demo, to explain what I mean. If the result is good enough I might try a PR even.

andig, 2025-09-08:
> No matter what target time I set in my charge plan to, if there is cheap power right after that, it will not be used, because the plan is designed to always meet its goal. But I would be happy to wait sometimes.

Not sure I understand. If you're happy to wait- just set a later charge goal? 

> For example, if my target time is 8am, and I plug in the car in the evening, the car will charge overnight even though there is cheap power at noon and I am happy with my min-charge being applied. 

You are aware that you can have multiple plans?

KonstantinSchubert, 2025-09-08:
> You are aware that you can have multiple plans?

Yes

> Not sure I understand. If you're happy to wait- just set a later charge goal?

 I set the charge goal for every evening at 8pm, I have the same problem that I have with 8am - if the car isn't full at 6pm, it will use whatever power available to charge it. 

What I really want is something that just charges my car when there is *relatively* cheap power. Like solar surplus charging, but there is no solar and you simply have time to *wait a bit* until the price curve hits the cheapest point *within the foreseeable future*.

andig, 2025-09-08:
> Something like "always plan the charge such that the car would reach X percentage in Y hours or less". 

That is **EXACTLY** the current planner. For minimum range use the minsoc on the arrival tab. That said, if you want a different algorithm you can always use the evcc APIs to implement your specific requirement externally.

KonstantinSchubert, 2025-09-08:
Hi, 

> That is EXACTLY the current planner. For minimum range use the minsoc on the arrival tab.

I agree. My proposal is basically the current planner, with the target time continuously moving forward.  I am open to ideas for better algorithms. 

If I just wanted to solve this for myself, I would be happy using the API or just edit the sqlite.db. But I think this might be useful for others too. 

Do you see the problem though that I am trying to solve?

People can set a min-soc, evcc will always fast-charge below that. And above that, people might not care if the car charges today, or tomorrow. A lot of people drive very little and they don't care if the car is at 80% at any fixed point in time - they are happy to wait a bit for more sun or cheaper power.

KonstantinSchubert, 2025-09-09:
One could think of a more complex algorithm, taking into account the predicted driving energy consumption of the car based on historical patterns, and then run a linear optimization to keep the charge of the car within a limited band, using the cheapest energy (pv and grid) available.

naltatis, 2025-09-09:
@KonstantinSchubert agreed. With the existing features (planner, fixed limits) we very deliberately prioritize predictability over cleverness/assumptions. So that the user is not surprise by the behavior of the system. In addition to this there is currently a lot of work going on in providing a smarter algorithm that can take actual usage and conflicting goals (home battery, car battery, ...) into account. You can search for the "Optimizer" here in GitHub or in Slack to find out more. Still early days but very promising.