evcc-io/evcc #23458: CHARX Phoenix charging issue

filiphaenen, 2025-09-08:
CHARX Phoenix issue;

My setup:

Solar System: GoodWe, PV + Battery; data and readings correct (although battery not separately inserted)

EV Charger: Phoenix Contact CHARX controller (VETON charger)

EVCC installed on separate server in docker, evcc is running correctly.
EVCC mode: min + PV

The issue I'm seeing is that charging starts, but the charger only goes to 16A constantly although evcc is requesting other currents (above 6A), logs attached.

Tested with another VETON charger and same behavior. It appears an error in the used template for CHARX, could that be a possibility?

[evcc-20250729-165420-trace.log](https://github.com/user-attachments/files/22114781/evcc-20250729-165420-trace.log)
[image]
[image]
[image]

thanks in advance

_Originally posted by @filiphaenen in https://github.com/evcc-io/evcc/discussions/21221#discussioncomment-14294344_

andig, 2025-09-08:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen. 

Please add a logfile, NO screenshots.

closed 2025-09-08 by andig

filiphaenen, 2025-09-10:
> > the charger only goes to 16A constantly although evcc is requesting other currents (above 6A), logs attached.
> 
> So everything is working as it should from evcc side?

That’s one way to see it? Before I used flexbox from lifepowr and that was able to control the charx properly. 

Does evcc use a specific template for charx or chargers?