evcc-io/evcc #23495: GUI Home Screen enable/disable parameter und lademodus Änderbarkeit

elektro-freak, 2025-09-10:
Hallo,

ich habe eine Wallbox und 2 Heizstäbe unter Laden & Heizen in der GUI konfiguriert.
Es wäre super wenn man in der Konfiguration je Wallbox/Heitzstab das editieren im Home Screen der Parameter sowie des Mode: Aus, PV etc. disabln könnte. Hintergrund ist recht simple. Eine Änderung an den Heizstäben muss man eigentlich nicht tätigen, da dies ja bei jedem auf PV stehen sollt und nicht verändert wird. Jedoch ändert man ja die Wallbox schonmal auf min+PV oder schnell. Wenn also z.B. ein Standard User (Familienmitglieder) mal schnell laden wollen, könnte eine kostspielig Fehlbedienung entstehen, Denn der Shelly disconnected nie und würde immer den Heitzstab mit Strom versorgen. Könnte also sein das dies auf dem Smartphone im Winter Wochenlang nicht auffällt.

siehe auch Discussion #23490 

Vielen Dank
Gruß

andig, 2025-09-10:
Aktuell ist kein differenziertes Benutzerkonzept vorgesehen.

sigma415, 2025-09-10:
> ...
Es gibt hier ein Proposal eine optimierte UI für Nicht-Fahrzeuge / einfachere Verbraucher / reine Zähler anzubieten. Unter der Überschrift werden wir das angehen. Das Thema lag, zugunsten von Config UI, schon relativ lange. Werden wir aber umsetzen.
> ...
> 
> https://github.com/evcc-io/evcc/discussions/7235

Endlich Mini-Loadpoints wäre wirklich super 😀