evcc-io/evcc #23582: Kathrein ac35 RFID wird nicht ausgelesen

skipperkm, 2025-09-14:
### Describe the bug

Carport: charger vehicle id: modbus: exception '2' (illegal data address), function '3'

### Steps to reproduce

1.

[image]

2.
3.
...

### Configuration details

```yaml
Mit ‚evcc charger‘ kann der Identifier nicht ermittelt werden. (Siehe Scrennshot)
```

### Log details

```text
site ] DEBUG 2025/09/14 10:49:01 ----
[lp-1 ] DEBUG 2025/09/14 10:49:01 charge power: 0W
[lp-1 ] DEBUG 2025/09/14 10:49:01 charge currents: [0.055 0.053 0.056]A
[lp-2 ] DEBUG 2025/09/14 10:49:01 charge power: 0W
[site ] DEBUG 2025/09/14 10:49:01 grid power: 4W
[site ] DEBUG 2025/09/14 10:49:01 grid currents: [0.96 0.48 0.96]A
[site ] DEBUG 2025/09/14 10:49:01 ext 1 power: -20W
[site ] DEBUG 2025/09/14 10:49:01 battery 1 power: -1289W
[site ] DEBUG 2025/09/14 10:49:01 pv 1 power: 1465W
[site ] DEBUG 2025/09/14 10:49:01 !! solar production: accumulate add pv1 0.000W power (was: Accumulated: 3730.332kWh updated: 2025-09-14 10:48:22 +0200 CEST)
[site ] DEBUG 2025/09/14 10:49:01 !! solar production: accumulate moved pv1 from 3730.332 to 3730.348
[site ] DEBUG 2025/09/14 10:49:02 battery 1 soc: 56%
[site ] DEBUG 2025/09/14 10:49:02 site power: -1035W
[lp-1 ] DEBUG 2025/09/14 10:49:02 vehicle odometer: 13096km
[lp-1 ] DEBUG 2025/09/14 10:49:02 charge voltages: [226 226 226]V
[lp-1 ] DEBUG 2025/09/14 10:49:02 charge total import: 332.855kWh
[lp-1 ] DEBUG 2025/09/14 10:49:02 charger status: B
[lp-1 ] ERROR 2025/09/14 10:49:02 charger vehicle id: modbus: exception '2' (illegal data address), function '3'
[site ] DEBUG 2025/09/14 10:49:02 solar forecast: accumulated -0.245Wh from 2025-09-14 10:48:22 +0200 CEST to 2025-09-14 10:49:02 +0200 CEST
[site ] DEBUG 2025/09/14 10:49:02 solar forecast: produced 3730.348
[site ] DEBUG 2025/09/14 10:49:02 solar forecast: accumulated 3409.382kWh, produced 3730.348kWh, scale 1.094
[site ] DEBUG 2025/09/14 10:49:41 ----
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.6

skipperkm, 2025-09-14:
Version 2.7.0

mfuchs1984, 2025-09-14:
Laut Photovoltaikforum soll man im "Web-Interface unter Einstellungen/Modbus TCP Steuerung die Register-Beschreibung runterladen" können. Kannst du da mal nachschauen und die Beschreibung hochladen?

skipperkm, 2025-09-14:
[Modbus-Server (1).pdf](https://github.com/user-attachments/files/22319130/Modbus-Server.1.pdf)

Hier die Beschreibung

mfuchs1984, 2025-09-14:
Hmm, das Register 70 ist drin. Kannst du einmal `evcc charger -- log trace` ausführen?

zomtec, 2025-09-21:
I‘m currently on the current nightly build of evcc and have the 2.7.0 Firmware on my (newly installed) AC-35.

Today I made the first charging test. I can see the charging data in evcc, but I still get the errors in the log. Also I cannot stop the charging via evcc:

````
[lp-1 ] DEBUG 2025/09/21 11:29:28 charge power: 7180W
[lp-1 ] DEBUG 2025/09/21 11:29:28 charge currents: [15.9 16 0.093]A
[lp-1 ] DEBUG 2025/09/21 11:29:28 charge voltages: [225 226 229]V
[lp-1 ] DEBUG 2025/09/21 11:29:28 detected active phases: 2p
[lp-1 ] DEBUG 2025/09/21 11:29:29 charge total import: 7.851kWh
[lp-1 ] DEBUG 2025/09/21 11:29:29 charger status: C
[lp-1 ] ERROR 2025/09/21 11:29:29 charger vehicle id: modbus: exception '2' (illegal data address), function '3'
[lp-1 ] ERROR 2025/09/21 11:29:29 charger enabled: read tcp 192.168.10.220:55530->192.168.10.222:502: read: connection reset by peer
``` `

zomtec, 2025-09-21:
Just for mentioning it: it’s the same with the latest nightly build: https://github.com/evcc-io/evcc/commit/dbb165eb

mfuchs1984, 2025-09-21:
Ich würde eher davon ausgehen, dass die Kathrein Firmware einen Fehler hat.

Dodi1968, 2025-09-21:
Danke für den naheliegenden Tipp mit dem Modbus-Tool. Ich denke, ich habe die Lösung gefunden. Es müssen nicht 16 sondern 24 Register ausgelesen werden. Das ist ein Fehler im Modbus-Dokument von Kathrein (ist nicht der erste).
Außerdem müssen noch die Null-Bytes am Ende und das Präfix "RFID:" entfernt werden. 
Mit dem Modbus-Tool konnte ich die RFID auslesen und auch den VOID-Fall nachvollziehen.
Einen entsprechenden PR habe ich bereits vorbereitet.

closed 2025-09-22 by andig

Dodi1968, 2025-09-26:
Das spricht aber auch für das Team von @kathrein, dass sie so schnell reagiert haben. Ich hatte den Fehler erst vor zwei Tagen per Ticket gemeldet.

PR #23758 (2025-09-22): Kathrein: fix RFID (part 2)
May fix #23582 

- Read 24 registers for RFID (instead of 16, error in Kathrein modbus document),
- Remove null bytes,
- Remove prefix “RFID:”