evcc-io/evcc #23600: EVCC starts charging immediately on startup, when there is no surplus

johnrogers, 2025-09-14:
### Describe the bug

Hi there,

I have been trying to get EVCC configured for some time now. I am getting close - for some reason when I start the container, it immediately starts charging, despite there not being any solar surplus (in fact, in this example there is actually current grid use, it should be nowhere near charging).

This is an interesting line from the logs:

```
[lp-1  ] DEBUG 2025/09/15 08:13:46 plan: charge 27h43m46s between 2025-09-15 08:13:46 +1000 AEST until 2025-09-16 06:20:00 +1000 AEST (overruns by 5h37m32s, power: 2300W, avg cost: 0.304)
```

I am not sure why it thinks this is the plan. For starters, the SoC is already higher than tomorrow morning's plan, I'm not sure why it would think that it needs to charge from now until tomorrow morning to reach the state of charge.

I am using the Tesla Fleet API to retrieve vehicle data, and TeslaBleHttpProxy to send commands via Bluetooth to the car. The bluetooth proxy works well. It appears that on application startup, EVCC will indicate that the car is below the minimum state of charge (30%) - what it seems to do is make assumptions about the state of the car without having data available. Perhaps this is why the above plan is configured? It thinks it is below 30% SoC so it has to charge for a long time. I am currently using a Tesla Mobile Connector at 10A so it stands to reason that this is the case (it would actually take that long).

It starts charging before the Vehicle's SoC is reported in the log:

```
[lp-1  ] DEBUG 2025/09/15 08:14:54 vehicle soc: 72%
```

Shouldn't it be waiting to figure out the SoC before it decides to start charging / come up with a plan?
Perhaps it is not waiting long enough for the Fleet API to indicate the car's state of charge before starting charging? It is definitely EVCC that starts the charge, not the car. The car had been plugged in for some time.

These are my configured plans:

[image]

The SoC is > 70%, so the next plan should not be tomorrow (16 September) morning at 6:20am. In fact, EVCC's own UI indicates that the next plan is on Saturday for 8am to get to 80%:

[image]

In order to address the issue, I have to manually stop the charge.

I have not seen this issue occur while EVCC has been running for a while, I just saw it on startup. This is not to say it doesn't happen, and I will keep monitoring.

### Steps to reproduce

1. Start the EVCC container
2. Observe that EVCC indicates that the car is below minimum SoC (<30%), even though the car is at 76%.
3. Observe that charging starts, and I have to manually stop it otherwise EVCC will not stop it.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
log: info
levels:
  cache: error

network:
  port: 7070
  host: evcc.local # .local suffix announces the hostname on MDNS

# unique installation id
plant: 67ed41dcdc8e42a1d5bb80b1f59516d6027e18c9adef1b0b58c8310462dc80ac

interval: 60s # control cycle interval

chargers:
  - name: my_charger
    type: template
    template: twc3
    host: 10.0.0.3:9090

meters:
  - type: template
    template: fronius-gen24 
    usage: grid
[... 153 lines cut ...]
[lp-1  ] INFO 2025/09/15 08:14:54 start charging ->
[lp-1  ] DEBUG 2025/09/15 08:14:54 wake-up timer: stop
[pushover] DEBUG 2025/09/15 08:14:54 sending to ***
[lp-1  ] DEBUG 2025/09/15 08:14:54 vehicle status: C
[lp-1  ] DEBUG 2025/09/15 08:14:54 vehicle soc: 72%
[lp-1  ] DEBUG 2025/09/15 08:14:54 vehicle soc limit: 80%
[lp-1  ] DEBUG 2025/09/15 08:14:54 vehicle range: 405km
[lp-1  ] DEBUG 2025/09/15 08:14:54 plan: charge 3h10m9s between 2025-09-20 02:49:51 +1000 AEST until 2025-09-20 07:00:00 +1000 AEST (power: 2300W, avg cost: 0.080)
[lp-1  ] DEBUG 2025/09/15 08:14:54 plan: continuing until end of slot at 2025-09-15 09:00:00 +1000 AEST
[site  ] DEBUG 2025/09/15 08:14:54 solar forecast: accumulated 0.039Wh from 2025-09-15 08:13:46 +1000 AEST to 2025-09-15 08:14:54 +1000 AEST
[site  ] DEBUG 2025/09/15 08:14:54 solar forecast: produced 750.923
[site  ] DEBUG 2025/09/15 08:14:54 solar forecast: accumulated 697.523kWh, produced 750.923kWh, scale 1.077
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

johnrogers, 2025-09-16:
> [@johnrogers](https://github.com/johnrogers) I tried this in my local dev and get the same error
> 
> try
> 
> ```
>       charging:
>         source: mqtt
>         topic: teslamate/cars/1/charger_actual_current
>         timeout: 1h
>         jq: . > 0
> ```
> 
> so using actual current as a value greater than 0 to indicate charging, I suspect because the string is `Stopped` instead of `"Stopped"` it's not valid json, but a plain number is valid.

@jeffborg, thank you so much!! Really appreciate your help.

I just tried that, and tested by putting EVCC into "Fast" mode, and it started the car charging correctly. Hurray!

When using MQTT, is it normal to get a lot of "outdated" messages? Every interval (currently 45 seconds), I get an outdated message for either `charger status` or `vehicle soc limit`:

```
[lp-1  ] ERROR 2025/09/16 16:12:30 charger status: outdated
[lp-1  ] ERROR 2025/09/16 16:13:15 vehicle soc limit: outdated
[lp-1  ] ERROR 2025/09/16 16:14:00 charger status: outdated
[lp-1  ] ERROR 2025/09/16 16:14:45 charger status: outdated
[lp-1  ] ERROR 2025/09/16 16:15:30 charger status: outdated
```

It seems to be reporting that the charger status is outdated every time it updates, but vehicle soc outdated disappears after one interval.

jeffborg, 2025-09-16:
@johnrogers not sure, I know there is some funky behavior with mqtt, I'm not sure if there is a http api for teslamate to get the same info. My setup it's the charger being controlled not the car. The car is really only there for it's SOC.

according to https://docs.evcc.io/en/docs/devices/plugins#mqtt the timeout is the max age of a message, plugged_in wouldn't be being pushed very often, it may need a 12/24h timeout.

johnrogers, 2025-09-16:
> [@johnrogers](https://github.com/johnrogers) not sure, I know there is some funky behavior with mqtt, I'm not sure if there is a http api for teslamate to get the same info. My setup it's the [...]

Yeah, fair enough. Thanks for all of your contributions, you are a legend. My EVCC seems to be working now.
I'll check out how it goes tomorrow with the sun out, and with a scheduled charge to end tomorrow morning to triple check.

Thanks a lot!!

LeonSol, 2025-09-16:
@johnrogers when you get messages like this:

[lp-1  ] ERROR 2025/09/16 16:12:30 charger status: outdated
[lp-1  ] ERROR 2025/09/16 16:13:15 vehicle soc limit: outdated
[lp-1  ] ERROR 2025/09/16 16:14:00 charger status: outdated
[lp-1  ] ERROR 2025/09/16 16:14:45 charger status: outdated
[lp-1  ] ERROR 2025/09/16 16:15:30 charger status: outdated

have a look at this article to get rid of it:

https://emqx.medium.com/the-beginners-guide-to-mqtt-retained-messages-a2b6cae6ab48

johnrogers, 2025-09-16:
> [@johnrogers](https://github.com/johnrogers) you have a charge at 8c always set. Smart cost. Turn that off. Evcc did the right thing.

@jeffborg thanks, how do I do that? I specified the tariffs in my config:

```
tariffs:
  currency: AUD
  grid:
    type: fixed
    price: 0.36619 # AUD/kWh
    zones:
      - days: Mon-Sun
        hours: 0-6
        price: 0.08 # AUD/kWh

  feedin:
    type: fixed
    price: 0.04 # AUD/kWh

  solar:
    - type: template
      template: forecast-solar
      lat: ***
      lon: ***
      dec: 20
      kwp: 5.0
      az: 135
```

I don't remember turning anything called smart cost on? Is it supposed in the UI some where that I've missed?

jeffborg, 2025-09-16:
@johnrogers see https://docs.evcc.io/en/docs/features/dynamic-prices#cheap-grid-charging

jeffborg, 2025-09-17:
@johnrogers not really technically it's charging, I think it's right for evcc to show it as charging because it is drawing power. Turn off preconditioning because if the car has power available it will draw from that. You need a smart charger or a add in a switched socket to cut power to prevent this.