evcc-io/evcc #23630: V2C TRYDAN_PRO does not work with OCPP

grobihh, 2025-09-16:
### Describe the bug

I am using the template "trydan" for my charger "V2C TRYDAN_PRO" but that does not work well.
Instead of controlling the charger (PV, Min+PV, Fast) I can only switch it on or off. The charger then follows its profiles.
That is very uncomfortable.

So I thought, no problem, the charger is OCPP capable.
When switching to OCPP for EVCC and the charger I get some OCPP errors and cannot start the charging session.

### Steps to reproduce

1. Change evcc config to use the ocpp template for the charger
2. Change the charger settings to use OCPP.
3. Restart the charger
4. Wait for charger to boot up again.
5. Charger confirms OCPP
6. Restart EVCC
7. => OCPP error
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: myserver.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: warn
  site: info
  mqtt: info
  lp-1: info
#  ocpp: trace

# unique installation id
plant: 81ba959084bbb27f70cafab3b3a95097409bf45de725c7b520efb9a7d2c5074c

interval: 30s # control cycle interval

meters:
- name: solar
  type: custom
  power: # power (W)
    source: mqtt
    topic: envoy/699/state
    jq: .instant
    timeout: 50000s # 14h to survive the night
- name: grid
  type: custom
  power: # power (W)
    source: mqtt
    topic: envoy/701/state
    jq: .instant
    timeout: 60s

chargers:
- name: v2c_trydan
  type: template
  host: V2C-Trydan
  template: ocpp
  stationid: trydan01

vehicles:
- name: car_offline
  type: template
  template: offline
  capacity: 32
[... 92 lines cut ...]
[ocpp ] TRACE 2025/09/02 12:34:48 recv trydan01: [2,"531531536","Heartbeat",{}]
[ocpp ] TRACE 2025/09/02 12:34:48 send trydan01: [3,"531531536",{"currentTime":"2025-09-02T10:34:48Z"}]
[ocpp ] TRACE 2025/09/02 12:35:17 send trydan01: [2,"3477757560","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2025/09/02 12:35:17 recv trydan01: [3,"3477757560",{"status":"Rejected"}]
[ocpp ] TRACE 2025/09/02 12:35:47 send trydan01: [2,"1238819108","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2025/09/02 12:35:47 recv trydan01: [3,"1238819108",{"status":"Rejected"}]
[ocpp ] TRACE 2025/09/02 12:35:48 recv trydan01: [2,"531531537","Heartbeat",{}]
[ocpp ] TRACE 2025/09/02 12:35:48 send trydan01: [3,"531531537",{"currentTime":"2025-09-02T10:35:48Z"}]
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.5

grobihh, 2025-09-16:
Btw. I believe this PR https://github.com/evcc-io/evcc/pull/22880 fixed EVCC crashing on start when EVCC failed to connect to my charger via OCCP.

andig, 2025-09-16:
> when EVCC failed to connect to my charger via OCCP

In Ocpp the charger connects to evcc.

closed 2025-09-16 by andig

premultiply, 2025-09-16:
Your charger reports that its firmware does not support the mandatory OCPP SmartCharging profile.

```json
    {
      "key": "SupportedFeatureProfiles",
      "readonly": false,
      "value": "Core"
    }
```