evcc-io/evcc #23642: EVCC Homeassistant Switch

cookingeek, 2025-09-16:
### Describe the bug

When the homeassistant-switch charger template is configured, EVCC continuously issues unauthenticated requests to Home Assistant (GET /api/states/...) and logs a repeating 404: Not Found.
The component keeps polling, causing log spam and confusion. E

1.	Configure a charger using the homeassistant-switch template without providing token.
2.	Start EVCC with -l trace.
3.	Observe repeated requests to Home Assistant and repeating 404 lines.

`chargers:
  - name: knx_steckdose
    type: template
    template: homeassistant-switch
    baseurl: http://homeassistant.local:8123
    token: <token>
    switchentity: switch.haustur_steckdose
    standbypower: 15`

`
[ha-switch] TRACE 2025/09/16 23:27:44 GET http://homeassistant.local:8123/api/states/
[ha-switch] TRACE 2025/09/16 23:27:44 404: Not Found
[ha-switch] TRACE 2025/09/16 23:27:44 GET http://homeassistant.local:8123/api/states/
[ha-switch] TRACE 2025/09/16 23:27:44 404: Not Found
(repeats…)
`

### Steps to reproduce

1.	Configure a charger using the homeassistant-switch template without providing token.
2.	Start EVCC with -l trace.
3.	Observe repeated requests to Home Assistant and repeating 404 lines.

...

### Configuration details

```yaml
chargers:
  - name: knx_steckdose
    type: template
    template: homeassistant-switch
    baseurl: http://homeassistant.local:8123
    token: <token>
    switchentity: switch.haustur_steckdose
    standbypower: 15
```

### Log details

```text
[ha-switch] TRACE 2025/09/16 23:27:44 GET http://homeassistant.local:8123/api/states/
[ha-switch] TRACE 2025/09/16 23:27:44 404: Not Found
[ha-switch] TRACE 2025/09/16 23:27:44 GET http://homeassistant.local:8123/api/states/
[ha-switch] TRACE 2025/09/16 23:27:44 404: Not Found
(repeats…)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

cookingeek, 2025-09-17:
Token is installed.

spuky, 2025-09-17:
you might need quotes around the token..

spuky, 2025-09-17:
and also provide success during config..
https://github.com/user-attachments/assets/d0720abf-ae53-4152-9336-9bf4997ac165

cookingeek, 2025-09-17:
I did it first via gui and then again with yaml configuration. Both have mainly the same issue.

spuky, 2025-09-17:
For the moment the best Option is to fake the Power Entity in HA by creating an template helper...

Example for s device that pulls 300W when on :

`{{ 300 if is_state('switch.examledevice', 'on') else 0 }}`