evcc-io/evcc #23643: Compatibility with Deye Inverters

tg-all, 2025-09-17:
E.g. Deye Hybrid-Wechselrichter SUN-12K-SG04LP3 12 kW 3 Phasen 48V

Hello,

Deye inverters are currently not directly supported by evcc over TCP IP.

Is it right?

I do not get any connection.

Greetings
Thomas

[image]
[image]

spuky, 2025-09-17:
does only work with an adapter see german discussion: https://github.com/evcc-io/evcc/discussions/22046

tg-all, 2025-09-17:
Der Homeassistant greift aktuell alle Daten ohne Waveshare per TCP IP.über den Logger ab. Die Daten sind irgendwie abrufbar. Wieso kann das der evcc nicht?

spuky, 2025-09-17:
Du kannst aber auch deine werte aus HA über MQTT schon bereitstellen... bis jemand ein Template dafür schreibt..

spuky, 2025-09-17:
> Vielen Dank für die Info und weiherhin viele Erfolg.

@tg-all das klingt so nach aufgeben...

Warum... evtl. teilst du mal den Link zur HA integration die du benutzt... daraus lässt sich dann ein Template ableiten das du testen kannst... und wenn es funktioniert können wir den wir auch über http unterstützen..

spuky, 2025-09-17:
@StefanSchoof sein logger macht modbus... und wenn er das solarman plugin konfiguriert sollte das laufen.. ;-)

tg-all, 2025-09-17:
Ich habe zwei Srceenshots dran.

[image]
[image]

tg-all, 2025-09-17:
Weiterer Screenshot

[http___192.168.178.10_status.html](https://github.com/user-attachments/files/22388460/http___192.168.178.10_status.html)

tg-all, 2025-09-17:
Im HO werden sehr viel mehr Daten angezeigt, als rein im Logger.  Wäre in der Yaml-Datei p3_deye.yaml zu erkennen, wie der HO ausliest? Welches Protokoll, welcher Port. 

Anbei Beispieldaten vom HO

[image]

tg-all, 2025-09-17:
Internationale WhatsApp Chatgruppe mit Deye Support Team:

https://chat.whatsapp.com/C4J1T82oka2IgXSfLU7mgd?mode=ems_copy_t

tg-all, 2025-09-17:
> Ich hab das html auf einen Testwebserver auf meiner lokalen maschine gepackt nach dem ich gesehen habe die Regex passen zum html..
> 
> der basic auth user admin und pw admin gepackt und bekomme...
> 
> bei Auswahl von IGEN Tech Solarman Logger das...
> 

Ich werd' verrückt, mit dem IGEN Tech Solarman Logger geht Energie und Leistung.

Interssant wäre natürlich Akku Ladezustand. Aus den ganzen Modbus Daten. Aber schon mal nicht schlecht 😀

[image]