evcc-io/evcc #23670: Enable Dutch CO2 forecast data from ned.nl

tbiest, 2025-09-18:
**Is your feature request related to a problem? Please describe.**
Just like green-grid-compass for DE I would like to use the CO2 forecast data from ned.nl api to charge when the CO2 intensity is lowest. I have the API description from ned.nl and looked into the existing template from green-grid-compass but I'm not sure it matches and I'm not sure if we need some custom coding for this.

**Describe the solution you'd like**
Feed the ned.nl co2 forecast data in evcc the regular way

**Describe alternatives you've considered**
There is no alternative

**Additional context**
Here is the request and response I think we need:
https://api.ned.nl/v1/utilizations?point=0&type=27&granularity=5&granularitytimezone=1&classification=1&activity=1&validfrom[strictly_before]=2025-09-19&validfrom[after]=2025-09-18
And you need to pass your own API key in the header (X-AUTH-TOKEN)

The response you get is as follows. Per hour you get the production co2 intensity. The emission factor is kg/kwh. You will need to multiply by 1000 to get grams/kwh. Timezone is in UTC, I think Amsterdam is +2 so that needs to be added.

To me it is not clear how evcc will pass this data into the correct data model.

[
    {
        "id": 70667476442,
        "point": "/v1/points/0",
        "type": "/v1/types/27",
        "granularity": "/v1/granularities/5",
        "granularitytimezone": "/v1/granularity_time_zones/0",
        "activity": "/v1/activities/1",
        "classification": "/v1/classifications/1",
        "capacity": 12299354,
        "volume": 12299354,
        "percentage": 0.29002851247787476,
        "emission": 591726,
        "emissionfactor": 0.048111073672771454,
        "validfrom": "2025-09-17T22:00:00+00:00",
        "validto": "2025-09-17T23:00:00+00:00",
        "lastupdate": "2025-09-17T23:51:01+00:00"
    },
    {
        "id": 70667476443,
        "point": "/v1/points/0",
        "type": "/v1/types/27",
        "granularity": "/v1/granularities/5",
        "granularitytimezone": "/v1/granularity_time_zones/0",
        "activity": "/v1/activities/1",
        "classification": "/v1/classifications/1",
        "capacity": 12078946,
        "volume": 12078946,
        "percentage": 0.284828245639801,
        "emission": 578856,
        "emissionfactor": 0.04792322590947151,
        "validfrom": "2025-09-17T23:00:00+00:00",
        "validto": "2025-09-18T00:00:00+00:00",
        "lastupdate": "2025-09-17T23:51:01+00:00"
    },
    {
        "id": 70667476444,
        "point": "/v1/points/0",
        "type": "/v1/types/27",
        "granularity": "/v1/granularities/5",
        "granularitytimezone": "/v1/granularity_time_zones/0",
        "activity": "/v1/activities/1",
        "classification": "/v1/classifications/1",
        "capacity": 12016713,
        "volume": 12016713,
        "percentage": 0.28335800766944885,
        "emission": 572384,
        "emissionfactor": 0.04763232544064522,
        "validfrom": "2025-09-18T00:00:00+00:00",
        "validto": "2025-09-18T01:00:00+00:00",
        "lastupdate": "2025-09-18T06:51:52+00:00"
    },
[... 333 lines cut ...]
        "percentage": 0.34931474924087524,
        "emission": 2601636,
        "emissionfactor": 0.17554624378681183,
        "validfrom": "2025-09-18T20:00:00+00:00",
        "validto": "2025-09-18T21:00:00+00:00",
        "lastupdate": "2025-09-18T06:51:52+00:00"
    },
    {
        "id": 70667858862,
        "point": "/v1/points/0",
        "type": "/v1/types/27",
        "granularity": "/v1/granularities/5",
        "granularitytimezone": "/v1/granularity_time_zones/0",
        "activity": "/v1/activities/1",
        "classification": "/v1/classifications/1",
        "capacity": 12294905,
        "volume": 12294905,
        "percentage": 0.28985899686813354,
        "emission": 1403875,
        "emissionfactor": 0.11418475210666656,
        "validfrom": "2025-09-18T21:00:00+00:00",
        "validto": "2025-09-18T22:00:00+00:00",
        "lastupdate": "2025-09-18T06:51:52+00:00"
    }
]

andig, 2025-09-18:
Happy to take a PR for a new template!

tbiest, 2025-09-22:
I see the pull request is merged already. If there are any technical issues left we will make a new PR to fix. For now I think we can close the issue.

closed 2025-09-22 by tbiest

rjbaat, 2025-09-27:
met trace enabled it shows this for the co2 domain. 

```
```
[co2 ] TRACE 2025/09/27 15:24:09 GET https://api.ned.nl/v1/utilizations?point=0&type=27&granularity=5&granularitytimezone=0&classification=1&activity=1&validfrom[strictly_before]=2025-09-27&validfrom[after]=2025-09-29
[co2 ] TRACE 2025/09/27 15:24:09 {"@context":"\/v1\/contexts\/Error","@type":"hydra:Error","hydra:title":"An error occurred","hydra:description":"Date filter must be chronological correct"}
```

tbiest, 2025-09-27:
I think the valid from and strictly before dates are unfortunately swapped in the last version of the template. When testing the URL with postman and changing the value "validfrom[strictly_before]" to "2025-09-29" and "validfrom[after] to "2025-09-27" I get the 200 - OK.

I also get some values from today (history). I don't know if this will break evcc but it would be good practice to filter them out in the jq with some additional statements.

tbiest, 2025-09-30:
Hi Gerrit Jan, yes this should be the solution for at least the bad request problem. I made a PR that was merged already so it should be solved in the next release.

A potential next problem will be that the api output contains some historical data that we might need to filter out with the JQ statement. But we will see.

Fijne dag!

tbiest, 2025-10-07:
@gerritjandebruin it should be working now with the latest evcc release. I can see the forecast in the GUI and in the meantime we also took care of a PR for a 15 minute interval.

There is one problem left: I see that the conversion from UTC to local time in the JQ statement is not working. This causes the GUI to show the data 2 hours later than it is supposed to be. Simply subtracting two hours from the UTC is not a good solution since we have winter/summer time in NL. Any ideas on how we can fix this?

gerritjandebruin, 2025-10-07:
[image]

I can confirm that it is now two hours off. 

ChatGPT suggests that `(.validfrom | strptime("%Y-%m-%dT%H:%M:%S%z") | mktime | strflocaltime("%FT%TZ"))` should be `(.validfrom | strptime("%Y-%m-%dT%H:%M:%S%z") | mktime | strflocaltime("%FT%T"))` but I have still no dev env setup. So will try later to see whether this makes sense.

tbiest, 2025-10-15:
@gerritjandebruin I updated to the latest evcc version and now the timing seems to be ok.