evcc-io/evcc #23690: jq templates can't deal with plain strings

jeffborg, 2025-09-19:
As referenced here and was a bit unexpected

If a mqtt topic contains json like or a plain number which is also json then jq can be used to do something with it. But if it's a string eg `Charging` because it's not quoted in the mqtt message then jq can't be used.

Wondering if something can be done to remedy this? The jq is very powerful and if somehow this could be quoted in via another pipe/plugin added, if the string could be converted from `Charging` to `"Charging"` then iq will be able to process it

> [@johnrogers](https://github.com/johnrogers) the error is that the returned string isn't a "booleanish" value. You can use jq to manipulate it to be one.
> 
> ```
>  charging:
>       source: mqtt
>       topic: teslamate/cars/1/charging_state
>       timeout: 60s
>       jq: . == "Charging" # should work untested
> ```
> 
> My value is Charging, I assume Disconnected is also valid. I think this needs `jq: . == "Charging"`
> 
> Using the data from teslamate directly will totally eliminate the 0% problem.
> 
> Thanks for the suggestion.
> 
> With the following:
> 
> ```
> status:
>     source: combined
>     plugged:
>       source: mqtt
>       topic: teslamate/cars/1/plugged_in
>       timeout: 60s
>     charging:
>       source: mqtt
>       topic: teslamate/cars/1/charging_state
>       timeout: 60s
>       jq: . == "Charging"
> ```
> 
> I get the error:
> 
> ```
> [lp-1  ] ERROR 2025/09/16 15:47:17 charger status: invalid character 'S' looking for beginning of value
> ```
> 
> Presumably this is something to do with the current `charging_state` value being `"Stopped"`. I'm just trying to figure out the right syntax... 

 _Originally posted by @johnrogers in [#23600](https://github.com/evcc-io/evcc/issues/23600#issuecomment-3295568841)_

spuky, 2025-09-19:
probier mal das als raw value zu parsen..

` jq: -R '. == "Charging"' `

andig, 2025-09-19:
Wow, great answer by claude! Here's what it does:

	if p.jq != nil {
		if p.quote {
			b = []byte(fmt.Sprintf("%q", string(b)))
		}
		v, err := jq.Query(p.jq, b)
		if err != nil {
			return b, backoff.Permanent(err)
		}
		b = []byte(fmt.Sprintf("%v", v))
	}

jeffborg, 2025-09-19:
@andig would be helpful if this was in the docs at https://docs.evcc.io/en/docs/devices/plugins#http no mention of quote option on the page.

closed 2025-09-20 by premultiply