evcc-io/evcc #23727: DaheimLaden Smart Pro - kurzzeitig 0kw trotz Überschuss

DirkDerris, 2025-09-20:
### Describe the bug

Wallbox: DaheimLaden Smart Pro 11kw
evcc: auf Raspberry Pi
Automatischer Wechsel / Phasenumschaltung: aktiviert

Siehe Screenshot der Huawei FusionSolar App - die Wallbox regelt teilweise auf 0kw obwohl noch genügend Überschuss vorhanden ist.

[evcc-20250920-173521-debug.log](https://github.com/user-attachments/files/22442444/evcc-20250920-173521-debug.log)
[evcc-20250920-173526-trace.log](https://github.com/user-attachments/files/22442443/evcc-20250920-173526-trace.log)

[image]
[image]

### Steps to reproduce

1. Wallbox-Steuerung über evcc
2. BMW-Auto an Wallbox anschließen
3. PV-Modus bei evcc aktivieren
4. Automatischer Wechsel / Phasenumschaltung: aktiviert

### Configuration details

```yaml
evcc.yaml Datei leer, da Konfiguration über UI
UI mit Standard-Einstellung (nur Wallbox, BMW-Auto und Huawei Emma-Smartmeter) hinzugefügt.
```

### Log details

```text
siehe beigefügte Dateien
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.2075

andig, 2025-09-21:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-09-21 by andig

VolkerK62, 2026-04-20:
Da gibt es sämliche Variationen.
Manche nein, manche ja, manche einstellbar.