evcc-io/evcc #23733: Wallbox Pulsar Plus - Edit the evcc.db and insert the command Remotestart = true

MisterFarmer73, 2025-09-21:
### Describe the bug

Hello,

has anyone a solution how I can edit the evcc.db to insert the needed command for my Wallbox Pulsar Plus ?
My Wallbox din´t accept the unlock command and so the loading will not start.

Or does anybody knows a solution for this probelem ?

I have a mac and a raspberry Pi 5 for running my EVCC

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2025/09/21 11:06:28 config file not found, database-only mode
[main  ] FATAL 2025/09/21 11:06:28 
Found systemd service database at "/var/lib/evcc/evcc.db", evcc has been invoked with no explicit database path.
Running the same config with multiple databases can lead to expiring vehicle tokens.

If you want to use the existing service database run the following command:

sudo evcc --database /var/lib/evcc/evcc.db

If you want to create a new user-space database run the following command:

evcc --database ~/.evcc/evcc.db

If you know what you're doing, you can skip the database check with the --ignore-db flag.
```

### Log details

```text
no log existing
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-09-21:
/cc @mfuchs1984

mfuchs1984, 2025-09-21:
`remotestart: true' is configurable in the same, is it missing in the UI?

MisterFarmer73, 2025-09-24:
> mfuchs1984
>  left a comment
> (evcc-io/evcc#23733)
> remotestart: true is configurable in the yaml, is it missing in the UI?

Dear Mr. Fuchs,

will you do an implentation the function in the near future or what can I do ?

Best Regards
Patrick

mfuchs1984, 2025-09-24:
I did not do anything, so does it mean it is already available in the UI configuration?