evcc-io/evcc #23734: Phase switch not stable anymore for Webasto Unite

ambitzgi, 2025-09-21:
### Describe the bug

From time to time the switch between 1 and 3 phases does not function any more.
Wallbox: Webasto Unite with most recent Vestel firmware 3.187.0
A few evcc versions before (I guess 0.205) it was working perfectly. 
After a reboot of the wallbox and restart of evcc it is working again. After some time (typically one day) the phase switching is not working anymore. If this is the case, in the web interface under the item "Ladestrom / Phasen" only two item are display ("3-phasig" & "1-phasig"). The third one (automatische Umschaltung) is then missing. If the switching is working again (after the above mentioned reboot / restart), the third item reappears. 

In the log see WARN at Sept 21 10:26:50  charger logic error: phases mismatch (got 1, expected 3)

### Steps to reproduce

1. Wait some time, typically one day,

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.78  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.78  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- type: template
  template: kostal-ksem-inverter 
  id: 71  
  host: 192.168.178.78  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid3

chargers:
- type: template
  template: vestel 
  id: 255  
  host: 192.168.178.103  
  port: 502  
  modbus: tcpip  
[... 126 lines cut ...]
Sep 21 10:26:50 openhabian4 evcc[2895077]: [lp-1  ] DEBUG 2025/09/21 10:26:50 available power 6441W > 4140W min 3p threshold
Sep 21 10:26:50 openhabian4 evcc[2895077]: [lp-1  ] DEBUG 2025/09/21 10:26:50 start phase scale3p timer
Sep 21 10:26:50 openhabian4 evcc[2895077]: [lp-1  ] DEBUG 2025/09/21 10:26:50 phase scale3p in 1m0s
Sep 21 10:26:50 openhabian4 evcc[2895077]: [lp-1  ] DEBUG 2025/09/21 10:26:50 pv charge current: 29.3A = 16A + 13.3A (-3050W @ 1p)
Sep 21 10:26:50 openhabian4 evcc[2895077]: [site  ] DEBUG 2025/09/21 10:26:50 solar forecast: accumulated 0.015Wh from 2025-09-21 10:26:30 +0200 CEST to 2025-09-21 10:26:50 +0200 CEST
Sep 21 10:26:50 openhabian4 evcc[2895077]: [site  ] DEBUG 2025/09/21 10:26:50 solar forecast: produced 4121.766
Sep 21 10:26:50 openhabian4 evcc[2895077]: [site  ] DEBUG 2025/09/21 10:26:50 solar forecast: accumulated 3045.040kWh, produced 4121.766kWh, scale 1.354
Sep 21 10:27:10 openhabian4 evcc[2895077]: [site  ] DEBUG 2025/09/21 10:27:10 ----
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.6

ambitzgi, 2025-09-21:
[image]

ambitzgi, 2025-09-21:
Question: What is the difference between „Soft Reset“ and „Hard Reset“?

ambitzgi, 2025-09-21:
Just one idea what could have triggered the issue. 
Withe the new evcc version I have changed the interval from 30s in there yaml to 20s in the web configuration interface. 
Right now I have changed back to 30s. Maybe this helps.

noi75, 2025-09-28:
Hi,

I have the same Problem. My Intervall was 30 sec all the time.
Is the Problem known by Vestel ?

Greeting Jan

wittj, 2026-02-04:
@ambitzgi I am running into the same problem. Did you find a result if the box is working stable with either 30s or daily reboot?

ambitzgi, 2026-11-16:
In the following issue a (potential) solution is described by upgrading to firmware 3.187.4 and afterwards to 3.187.42. 
https://github.com/evcc-io/evcc/discussions/33829
The download link for the firmware is provided too. 
I have installed the updates yesterday night. Everything is still working. However I cannot confirm that the phase switch problem is solved yet, because it is raining today.