evcc-io/evcc #23737: EVCC "imagines" more power then currently produced by PV

realwuffi, 2025-09-21:
### Describe the bug

EVCC uses more power for charging (in PV mode) than actually available.

This is reproducible, always when I start the charge process, I see the following behavior:

1. Current values of my Victron system (I have also checked the two inverters, the values match)
before starting the charging process and the values displayed in EVCC UI

[image]

[image]

2. After starting the charge process in "PV" mode

[image]

This seems to be a new issue, I have not seen this in EVCC before.
I have not charged within the last 10 days, so I believe it must have been introduced with the latest version (0.207.6).
But there has also been an update of the Victron system in the meantime, which could probably be the reason as well.

The chrge process started at 12:31:37

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
site:
  title: Daheim
  meters:
    grid: my_grid
    pv:
      - pv_all_victron
    battery:
      - pylontech_battery
  residualPower: 300

circuits:
  - name: main # if there is only one circuit defined the name needs to be 'main'
    title: 'main circuit'

# define your loadpoints according your needs
# see https://docs.evcc.io/en/docs/reference/configuration/loadpoints
loadpoints:
  - title: Sungrow Wallbox 
    charger: A24802C6839 
    vehicle: car_id3

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: my_grid
    type: template
    template: victron-energy 
    usage: grid
    host: 192.168.2.210
    meterid: 40
  - name: pv_all_victron
    type: template
    template: victron-energy 
    usage: pv
    host: 192.168.2.210 
  - name: pylontech_battery 
[... 709 lines cut ...]
[lp-1  ] DEBUG 2025/09/21 12:32:17 vehicle soc: 46%
[lp-1  ] DEBUG 2025/09/21 12:32:17 vehicle soc limit: 100%
[lp-1  ] DEBUG 2025/09/21 12:32:17 vehicle range: 171km
[lp-1  ] DEBUG 2025/09/21 12:32:17 pv charge current: 8.95A = 9.15A + -0.194A (134W @ 3p)
[lp-1  ] DEBUG 2025/09/21 12:32:17 set charge current limit: 8.95A
[site  ] DEBUG 2025/09/21 12:32:17 solar forecast: accumulated 0.022Wh from 2025-09-21 12:31:57 +0200 CEST to 2025-09-21 12:32:17 +0200 CEST
[site  ] DEBUG 2025/09/21 12:32:17 solar forecast: produced 1480.198
[site  ] DEBUG 2025/09/21 12:32:17 solar forecast: accumulated 1359.076kWh, produced 1480.198kWh, scale 1.089
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.6

andig, 2025-09-21:
Run

    evcc meter

and check which values are wrong.

realwuffi, 2025-09-21:
This is the output of EVCC meter:

/app # evcc -c /config/evcc.yaml meter
[main  ] INFO 2025/09/21 13:57:28 using config file: /config/evcc.yaml
[db    ] INFO 2025/09/21 13:57:28 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2025/09/21 13:57:28 connecting evcc-212715163 at tcp://192.168.2.101:1883
[mqtt  ] DEBUG 2025/09/21 13:57:28 tcp://192.168.2.101:1883 connected
pylontech_battery
-----------------
Power:        78W
Soc:          99%
Capacity:     9.6kWh
Controllable: true

pv_all_victron
--------------
Power: 4354W

my_grid
-------
Power:          -473W
Current L1..L3: -11.1A 3.5A 7.7A

This is what the UI shows

[image]

mfuchs1984, 2025-09-21:
Evcc fängt mit 6A an, erhöht dann aber nach und nach, da kein Netzbezug entsteht.

Deine grid power und die currents passen irgendwie auch nicht zusammen:

```
[site  ] DEBUG 2025/09/21 12:32:16 battery 1 power: -109W
[site  ] DEBUG 2025/09/21 12:32:16 grid power: -57W
[site  ] DEBUG 2025/09/21 12:32:16 grid currents: [-10.4 3.9 8.9]A
``
12,8 A gehen rein, 10,4 raus, macht bei 230V eigentlich 552 W Netzbezug.

1. Schau mal wie deine Residualleistung in der GUI (unter Regelverhalten) eingestellt ist
2. Erhöhe ggf. das interval auf 1 Minute (auch unter Regelverhalten)