evcc-io/evcc #23763: Crash when calling evcc meter

spuky, 2025-09-22:
### Describe the bug

Was collecting info for another Report:

[image]

```
root@evcc:/etc# evcc --database /var/lib/evcc/evcc.db meter
[main  ] INFO 2025/09/22 09:49:55 using config file: /etc/evcc.yaml
[db    ] INFO 2025/09/22 09:49:55 using sqlite database: /var/lib/evcc/evcc.db
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x50 pc=0x27f3428]

goroutine 1 [running]:
github.com/evcc-io/evcc/cmd.runMeter(0x67257c0, {0xc0004ed0c0, 0x0, 0x2})
        github.com/evcc-io/evcc/cmd/meter.go:74 +0x2c8
github.com/spf13/cobra.(*Command).execute(0x67257c0, {0xc0004ed0a0, 0x2, 0x2})
        github.com/spf13/cobra@v1.10.1/command.go:1019 +0xae7
github.com/spf13/cobra.(*Command).ExecuteC(0x6725aa0)
        github.com/spf13/cobra@v1.10.1/command.go:1148 +0x465
github.com/spf13/cobra.(*Command).Execute(...)
        github.com/spf13/cobra@v1.10.1/command.go:1071
github.com/evcc-io/evcc/cmd.Execute()
        github.com/evcc-io/evcc/cmd/root.go:124 +0x1a
main.main()
        github.com/evcc-io/evcc/main.go:44 +0x2d
```

### Steps to reproduce

1. call evcc meter from cli
2. Crash output
...

### Configuration details

```yaml
root@evcc:/etc# cat /etc/evcc.yaml 
 meters:
  - name: EV-Lader
    type: template
    template: shelly-pro-3em
    usage: ext
    host: 10.100.1.248 # IP-Adresse oder Hostname

root@evcc:/etc# evcc --database /var/lib/evcc/evcc.db dump -cfg
[main  ] FATAL 2025/09/22 09:56:39 failed reading config file: Unsupported Config Type ""

config from the GH ui screen:

{
  "site": {
    "title": "vg",
    "grid": "db:1",
    "pv": ["db:2"],
    "battery": ["db:3"],
    "aux": null,
    "ext": null
  },
  "loadpoints": [
    {"id":6,"name":"db:6","charger":"db:5","title":"Punkt 1","defaultMode":"off","priority":0,"phasesConfigured":3,"minCurrent":6,"maxCurrent":32,"smartCostLimit":null,"smartFeedInPriorityLimit":null,"planEnergy":0,"planTime":"0001-01-01T00:00:00Z","planPrecondition":0,"limitEnergy":0,"limitSoc":0,"thresholds":{"enable":{"delay":60000000000,"threshold":0},"disable":{"delay":180000000000,"threshold":0}},"soc":{"poll":{"mode":"charging","interval":3600000000000},"estimate":true}},
    {"id":9,"name":"db:9","charger":"db:8","title":"Punkt 2","defaultMode":"off","priority":0,"phasesConfigured":3,"minCurrent":6,"maxCurrent":32,"smartCostLimit":null,"smartFeedInPriorityLimit":null,"planEnergy":0,"planTime":"0001-01-01T00:00:00Z","planPrecondition":0,"limitEnergy":0,"limitSoc":0,"thresholds":{"enable":{"delay":60000000000,"threshold":0},"disable":{"delay":180000000000,"threshold":0}},"soc":{"poll":{"mode":"charging","interval":3600000000000},"estimate":true}}
  ],
  "charger": [
    {"config":{"connector":"1","connecttimeout":"5m","meterinterval":"10s","stationid":"TACW2242822T8451","template":"ocpp-abb-tac"},"deviceProduct":"ABB Terra AC (OCPP)","id":5,"name":"db:5","type":"template"},
    {"config":{"connector":"1","connecttimeout":"5m","meterinterval":"10s","stationid":"TACW2242822T8442","template":"ocpp-abb-tac"},"deviceProduct":"ABB Terra AC (OCPP)","id":8,"name":"db:8","type":"template"}
  ],
  "meter": [
    {"config":{"capacity":40,"host":"10.100.1.232","maxacpower":"0","password":"***","template":"openems","usage":"battery","watchdog":"60s"},"deviceProduct":"FENECON","deviceTitle":"Fenecon","id":3,"name":"db:3","type":"template"},
    {"config":{"host":"10.100.1.232","maxacpower":"0","password":"***","template":"openems","usage":"grid","watchdog":"60s"},"deviceProduct":"FENECON","id":1,"name":"db:1","type":"template"},
    {"config":{"host":"10.100.1.232","maxacpower":"0","password":"***","template":"openems","usage":"pv","watchdog":"60s"},"deviceProduct":"FENECON","deviceTitle":"Fenecon","id":2,"name":"db:2","type":"template"}
  ],
  "vehicle": [
    {"config":{"icon":"car","template":"offline","title":"Gastfahrzeug"},"deviceProduct":"Generisches Fahrzeug (ohne API)","id":4,"name":"db:4","type":"template"},
    {"config":{"capacity":10.4,"icon":"car","identifiers":["NO.000000024521"],"maxCurrent":20,"mode":"pv","phases":"1","template":"offline","title":"Ollis Passat"},"deviceProduct":"Generisches Fahrzeug (ohne API)","id":7,"name":"db:7","type":"template"},
    {"config":{"capacity":76,"icon":"car","identifiers":["NO.000000182270"],"maxCurrent":20,"mode":"pv","phases":"","template":"offline","title":"Philips Volvo"},"deviceProduct":"Generisches Fahrzeug (ohne API)","id":10,"name":"db:10","type":"template"}
  ],
  "tariffs": "#currency: EUR

grid: # price using energy from the grid
  type: fixed
  price: 0.35 # EUR/kWh

feedin: # price for feeding solar energy to the grid
  type: fixed
  price: 0.07 # EUR/kWh

co2: # carbon intensity forecast
  type: template
  template: grünstromindex
  zip: 89150
  token: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VyaWQiOiJzc3cycjExMHh4bG93MzYwamsybGsiLCJncm91cCI6ImFub255bW91cyIsImFwcGlkIjoiMHhGNTM3MDBmNDUwZTAzNTFkYjRjOWZGMUZlZkFmMTBFYjg1REEzQjcwIiwiZXhwIjoxNzU4NzE4NjU3LCJpYXQiOjE3NTU1MjE4NTd9.VcPpUV3rKDqD0G_-VdDFpd2LrZ4Pdm6nr0N-j1xVToM

# 48.49119396727466, 9.707371782332329  

solar:
    - type: template
      template: open-meteo
      lat: 48.4911 # Breitengrad
      lon: 9.7073 # Längengrad
      dec: 10 # Neigung, 0 = horizontal, 90 = vertikal
      kwp: 24.5 # Maximale Generatorleistung (kWp)
      az: -41 # Azimut, -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
    - type: template
      template: open-meteo
      lat: 48.4911 # Breitengrad
      lon: 9.7073 # Längengrad
      dec: 10 # Neigung, 0 = horizontal, 90 = vertikal
      kwp: 24.5 # Maximale Generatorleistung (kWp)
      az: 131 # Azimut, -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden"
}
```

### Log details

```text
no relevant logs available...
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.6 (687a507d)

spuky, 2025-09-22:
I added the ext meter as prepartion for loadmanagement but that is not configured yet....
evcc meter was showing the meter in previous versions in the CLI

andig, 2025-09-22:
Closed in 9f4e5f6b2b0c37cfb37c6f0db9cf4e7f3fe0a7b1

closed 2025-09-22 by andig

andig, 2025-09-22:
@naltatis im UI scheint das nicht zu funktionieren. Bei `redactMap` ist der Parameter drin:

```go
// configRedactSecrets defines keys that should be redacted from configuration files
var configRedactSecrets = []string{
	"mac",                   // infrastructure
	"sponsortoken", "plant", // global settings
	"apikey", "user", "password", "pin", // users
	"token", "access", "refresh", "accesstoken", "refreshtoken", // tokens, including template variations
	"ain", "secret", "serial", "deviceid", "machineid", "idtag", // devices
	"app", "chats", "recipients", // push messaging
	"vin",               // vehicles
	"lat", "lon", "zip", // solar forecast
}
```

Liegt das hier am yaml Konstrukt? Dann müssen wir das für die Zeit wohl mal aushalten...