evcc-io/evcc #23764: Better privatise Konfig output

spuky, 2025-09-22:
## Description

When submitting an Issue via evcc and sharing my config... I find the following info users might care about not sharing but might miss in the hurry..

## Steps to Reproduce

1 Submit a Bug report...

## Configuration (YAML)

```yaml
Failed to load configuration
```
not relevant here... probably Cloudflare filtering the request..

## Configuration (UI)

```json5
{
  "site": {
    "title": "vg",
    "grid": "db:1",
    "pv": ["db:2"],
    "battery": ["db:3"],
    "aux": null,
    "ext": null
  },
  "loadpoints": [
    {"id":6,"name":"db:6","charger":"db:5","title":"Punkt 1","defaultMode":"off","priority":0,"phasesConfigured":3,"minCurrent":6,"maxCurrent":32,"smartCostLimit":null,"smartFeedInPriorityLimit":null,"planEnergy":0,"planTime":"0001-01-01T00:00:00Z","planPrecondition":0,"limitEnergy":0,"limitSoc":0,"thresholds":{"enable":{"delay":60000000000,"threshold":0},"disable":{"delay":180000000000,"threshold":0}},"soc":{"poll":{"mode":"charging","interval":3600000000000},"estimate":true}},
    {"id":9,"name":"db:9","charger":"db:8","title":"Punkt 2","defaultMode":"off","priority":0,"phasesConfigured":3,"minCurrent":6,"maxCurrent":32,"smartCostLimit":null,"smartFeedInPriorityLimit":null,"planEnergy":0,"planTime":"0001-01-01T00:00:00Z","planPrecondition":0,"limitEnergy":0,"limitSoc":0,"thresholds":{"enable":{"delay":60000000000,"threshold":0},"disable":{"delay":180000000000,"threshold":0}},"soc":{"poll":{"mode":"charging","interval":3600000000000},"estimate":true}}
  ],
  "charger": [
    {"config":{"connector":"1","connecttimeout":"5m","meterinterval":"10s","stationid":"TACW2242822T8451","template":"ocpp-abb-tac"},"deviceProduct":"ABB Terra AC (OCPP)","id":5,"name":"db:5","type":"template"},
    {"config":{"connector":"1","connecttimeout":"5m","meterinterval":"10s","stationid":"TACW2242822T8442","template":"ocpp-abb-tac"},"deviceProduct":"ABB Terra AC (OCPP)","id":8,"name":"db:8","type":"template"}
  ],
  "meter": [
    {"config":{"capacity":40,"host":"10.100.1.232","maxacpower":"0","password":"***","template":"openems","usage":"battery","watchdog":"60s"},"deviceProduct":"FENECON","deviceTitle":"Fenecon","id":3,"name":"db:3","type":"template"},
    {"config":{"host":"10.100.1.232","maxacpower":"0","password":"***","template":"openems","usage":"grid","watchdog":"60s"},"deviceProduct":"FENECON","id":1,"name":"db:1","type":"template"},
    {"config":{"host":"10.100.1.232","maxacpower":"0","password":"***","template":"openems","usage":"pv","watchdog":"60s"},"deviceProduct":"FENECON","deviceTitle":"Fenecon","id":2,"name":"db:2","type":"template"}
  ],
  "vehicle": [
    {"config":{"icon":"car","template":"offline","title":"Gastfahrzeug"},"deviceProduct":"Generisches Fahrzeug (ohne API)","id":4,"name":"db:4","type":"template"},
    {"config":{"capacity":10.4,"icon":"car","identifiers":["NO.000000024521"],"maxCurrent":20,"mode":"pv","phases":"1","template":"offline","title":"Ollis Passat"},"deviceProduct":"Generisches Fahrzeug (ohne API)","id":7,"name":"db:7","type":"template"},
    {"config":{"capacity":76,"icon":"car","identifiers":["NO.000000182270"],"maxCurrent":20,"mode":"pv","phases":"","template":"offline","title":"Philips Volvo"},"deviceProduct":"Generisches Fahrzeug (ohne API)","id":10,"name":"db:10","type":"template"}
  ],
  "tariffs": "#currency: EUR

grid: # price using energy from the grid
  type: fixed
  price: 0.35 # EUR/kWh

feedin: # price for feeding solar energy to the grid
  type: fixed
  price: 0.07 # EUR/kWh

co2: # carbon intensity forecast
  type: template
  template: grünstromindex
  zip: 89150
  token: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VyaWQiOiJzc3cycjExMHh4bG93MzYwamsybGsiLCJncm91cCI6ImFub255bW91cyIsImFwcGlkIjoiMHhGNTM3MDBmNDUwZTAzNTFkYjRjOWZGMUZlZkFmMTBFYjg1REEzQjcwIiwiZXhwIjoxNzU4NzE4NjU3LCJpYXQiOjE3NTU1MjE4NTd9.VcPpUV3rKDqD0G_-VdDFpd2LrZ4Pdm6nr0N-j1xVToM

solar:
    - type: template
      template: open-meteo
      lat: 48.4911 # Breitengrad
      lon: 9.7073 # Längengrad
      dec: 10 # Neigung, 0 = horizontal, 90 = vertikal
      kwp: 24.5 # Maximale Generatorleistung (kWp)
      az: -41 # Azimut, -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
    - type: template
      template: open-meteo
      lat: 48.4911 # Breitengrad
      lon: 9.7073 # Längengrad
      dec: 10 # Neigung, 0 = horizontal, 90 = vertikal
      kwp: 24.5 # Maximale Generatorleistung (kWp)
      az: 131 # Azimut, -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden"
}
```

I can see the lat and long of my installation 
RFID IDs for Cars...
Grünstromindex token..
Station IDs of Chargers

## Logs

```
[lp-2  ] DEBUG 2025/09/22 10:41:28 phase timer inactive
[lp-2  ] DEBUG 2025/09/22 10:41:28 pv timer inactive
[site  ] DEBUG 2025/09/22 10:41:28 ----
[lp-2  ] DEBUG 2025/09/22 10:41:28 charge power: 0W
[lp-2  ] DEBUG 2025/09/22 10:41:28 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/09/22 10:41:28 charge power: 0W
[lp-1  ] DEBUG 2025/09/22 10:41:28 charge currents: [0 0 0]A
[site  ] DEBUG 2025/09/22 10:41:28 battery 1 power: 1219W
[site  ] DEBUG 2025/09/22 10:41:28 pv 1 power: 1392W
[site  ] DEBUG 2025/09/22 10:41:28 grid power: 109W
[site  ] DEBUG 2025/09/22 10:41:28 !! solar production: accumulate set db:2 29597.287kWh meter total (was: Accumulated: 5875.172kWh updated: 0001-01-01 00:00:00 +0000 UTC)
[site  ] DEBUG 2025/09/22 10:41:28 !! solar production: accumulate moved db:2 from 5875.172 to 5875.172
[site  ] DEBUG 2025/09/22 10:41:28 battery 1 soc: 51%
[site  ] DEBUG 2025/09/22 10:41:28 site power: 1528W
[lp-1  ] DEBUG 2025/09/22 10:41:28 charge voltages: [232 230 232]V
[lp-1  ] DEBUG 2025/09/22 10:41:28 detected connected phases: 3p
[lp-1  ] DEBUG 2025/09/22 10:41:28 charge total import: 34.237kWh
[lp-1  ] DEBUG 2025/09/22 10:41:28 charger status: A
[lp-1  ] INFO 2025/09/22 10:41:28 car disconnected
[lp-1  ] DEBUG 2025/09/22 10:41:28 set battery boost: false
[lp-1  ] DEBUG 2025/09/22 10:41:28 set session soc limit: 0
[lp-1  ] DEBUG 2025/09/22 10:41:28 set session energy limit: 0
[site  ] DEBUG 2025/09/22 10:41:29 solar forecast: accumulated 0.000Wh from 2025-09-22 10:41:28 +0200 CEST to 2025-09-22 10:41:29 +0200 CEST
[site  ] DEBUG 2025/09/22 10:41:29 solar forecast: produced 5875.172
[site  ] DEBUG 2025/09/22 10:41:29 solar forecast: accumulated 6360.453kWh, produced 5875.172kWh, scale 0.924
```

## Version

v0.207.6 (687a507d)

spuky, 2025-09-22:
By the way Love the Workflow..

andig, 2025-09-22:
> lat and long of my installation

Should be fixed in master

> RFID IDs for Cars...

Where?

> Grünstromindex token..

Where?

> Station IDs of Chargers

I wouldn't care about these.

spuky, 2025-09-22:
> > RFID IDs for Cars...
> 
> Where?
> 
> >
NO.000000024521
NO.000000182270

"vehicle": [
    "identifiers

```
"vehicle": [
    {"config":{"icon":"car","template":"offline","title":"Gastfahrzeug"},"deviceProduct":"Generisches Fahrzeug (ohne API)","id":4,"name":"db:4","type":"template"},
    {"config":{"capacity":10.4,"icon":"car","identifiers":["NO.000000024521"],"maxCurrent":20,"mode":"pv","phases":"1","template":"offline","title":"Ollis Passat"},"deviceProduct":"Generisches Fahrzeug (ohne API)","id":7,"name":"db:7","type":"template"},
    {"config":{"capacity":76,"icon":"car","identifiers":["NO.000000182270"],"maxCurrent":20,"mode":"pv","phases":"","template":"offline","title":"Philips Volvo"},"deviceProduct":"Generisches Fahrzeug (ohne API)","id":10,"name":"db:10","type":"template"}
  ],
```

> Where?
> 
> > Grünstromindex token..

ah there is also a zip

"tariffs":

```
co2: # carbon intensity forecast
  type: template
  template: grünstromindex
  zip: 89150
  token: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VyaWQiOiJzc3cycjExMHh4bG93MzYwamsybGsiLCJncm91cCI6ImFub255bW91cyIsImFwcGlkIjoiMHhGNTM3MDBmNDUwZTAzNTFkYjRjOWZGMUZlZkFmMTBFYjg1REEzQjcwIiwiZXhwIjoxNzU4NzE4NjU3LCJpYXQiOjE3NTU1MjE4NTd9.VcPpUV3rKDqD0G_-VdDFpd2LrZ4Pdm6nr0N-j1xVToM
```

> > Station IDs of Chargers
> 
> I wouldn't care about these.

Ok for me was on the fence when reporting...

naltatis, 2025-09-22:
Yes, we currently don't have a process for UI-configured tariffs since they use raw yaml.

Option 1: We could reuse the `evcc.yaml` redactor for yaml-response config endpoints. This would have to be opt-in (`?redact=true`) since the config UI needs unreacted values for load/edit/save cycles.

Option 2: Add a warning to the issue UI when tariff-yaml exists. This would alert the user do manually redact before submitting.

Option 3: Implement proper template based tariff configuration via UI (similar to meters, chargers, ...). This ways we don't need yaml (except for user-defined tariffs) and get proper redaction from out template/device mechanism (masked fields).

Last option is something we have to to anyways, but it involves a some work (concept and code) to do properly. For a short-term fix "Option 1" sounds like a good solution. The redact true param would only be set in the issues use-case.

andig, 2025-09-22:
Lets wait for 3...

naltatis, 2026-02-28:
Was fixed with https://github.com/evcc-io/evcc/pull/25039

closed 2026-02-28 by naltatis