evcc-io/evcc #23803: evcc try to charge when SoC is 100%

magicWF, 2025-09-24:
### Describe the bug

I use PV charging and change the session target to 100 % SoC. After the vehicle reached the 100% SoC, it stopped charging. But evcc try to start charge again. When I reduce the session target to 95% SoC, than all it fine: evcc recordnize that the SoC from vehicle is 100% > 95% session target SoC and stopp charging

### Steps to reproduce

1. start charging
2. set session target to 100% SoC
3. wait until vehicle has 100% SoC
4.  instead to stop charging evcc try to continue charging.
...

### Configuration details

```yaml
evcc control the VW e-UP with OVMS. The wallbox is a OpenWB Pro+.

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.home # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error

# unique installation id
plant: *****
# evcc Sponsortoken , valid until 2025-05-21 + 1095 days
sponsortoken: *****

interval: 30s # control cycle interval

influx:
  url: http://db01.home:8086
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  token: *****
  org: Home

mqtt:
  broker: mqtt.home:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: *****
  password: *****

meters:
- name: grid1
  #usage: grid
  type: custom
  power: # power (W)
    source: mqtt # plugin type
    # vz uuid: 38d3d680-5a14-11e8-af69-e70a9475f5fe
    topic: vzlogger/data/chn0/raw
    jq: .value
    # timeout:
    scale: 1
  # energy: # optional energy (kWh)
  #   source: mqtt # plugin type
  #   # vz uuid: 38d3d680-5a14-11e8-af69-e70a9475f5fe
  #   topic: vzlogger/data/chn0/raw
  #   jq: .value
  #   # timeout:
  #   scale: 1
  currents: # optional currents (A)
    - source: mqtt # L1 plugin type
      # vz uuid: 16ed5530-0a48-11ea-927f-a3d98f9396e1
[... 413 lines cut ...]
[lp-2  ] DEBUG 2025/09/24 07:09:26 charge power: 0W
[lp-2  ] DEBUG 2025/09/24 07:09:26 charge total import: 63.110kWh
[charger_openwb01_custom] TRACE 2025/09/24 07:09:26 GET http://ovms.dexters-web.de:6868/api/charge/WF-XX000/?username=XXXX&password=XxXxXxXxXxXxXxXxX
[charger_openwb01_custom] TRACE 2025/09/24 07:09:26 {"batt_capacity":"27.4","batt_current":"-0.00","batt_range_speed":"0.0","battvoltage":"349.50","cac100":"101.98","carawake":0,"caron":0,"charge_estimate":"0","charge_etr_full":"0","charge_etr_limit":"0","charge_etr_range":"0","charge_etr_soc":"0","charge_kwh_grid":"1.5","charge_kwh_grid_total":"8.0","charge_limit_range":"0","charge_limit_soc":"80","charge_timestamp":"-1","chargeb4":"0","chargecurrent":"0.00","chargeduration":"0","chargekwh":"12","chargelimit":"11","chargepower":"0.00","chargepowerinput":"0.00","chargerefficiency":"0.00","chargestarttime":"0","chargestate":"stopped","chargesubstate":"0","chargetimermode":"0","chargetimerstale":"0","chargetype":"0","charging":0,"charging_12v":0,"cooldown_active":"-1","cooldown_tbattery":"0","cooldown_timelimit":"0","cp_dooropen":4,"estimatedrange":"256","idealrange":"220","idealrange_max":"220","linevoltage":"0","m_msgage_d":84,"m_msgage_s":84,"m_msgtime_d":"2025-09-24 07:08:02","m_msgtime_s":"2025-09-24 07:08:02","mode":"range","pilotpresent":0,"soc":"100.0","soh":"84.98","staleambient":"1","staletemps":"1","temperature_ambient":"10","temperature_battery":"11.1094","temperature_cabin":"11","temperature_charger":"17","temperature_motor":"14.1719","temperature_pem":"18","units":"K","vehicle12v":"12.46","vehicle12v_current":"0","vehicle12v_ref":"12.54"}
[lp-2  ] DEBUG 2025/09/24 07:09:26 charger status: B
[charger_openwb01_custom] TRACE 2025/09/24 07:09:26 GET http://ovms.dexters-web.de:6868/api/charge/WF-XX000/?username=XXXX&password=XxXxXxXxXxXxXxXxX
[charger_openwb01_custom] TRACE 2025/09/24 07:09:26 {"batt_capacity":"27.4","batt_current":"-0.00","batt_range_speed":"0.0","battvoltage":"349.50","cac100":"101.98","carawake":0,"caron":0,"charge_estimate":"0","charge_etr_full":"0","charge_etr_limit":"0","charge_etr_range":"0","charge_etr_soc":"0","charge_kwh_grid":"1.5","charge_kwh_grid_total":"8.0","charge_limit_range":"0","charge_limit_soc":"80","charge_timestamp":"-1","chargeb4":"0","chargecurrent":"0.00","chargeduration":"0","chargekwh":"12","chargelimit":"11","chargepower":"0.00","chargepowerinput":"0.00","chargerefficiency":"0.00","chargestarttime":"0","chargestate":"stopped","chargesubstate":"0","chargetimermode":"0","chargetimerstale":"0","chargetype":"0","charging":0,"charging_12v":0,"cooldown_active":"-1","cooldown_tbattery":"0","cooldown_timelimit":"0","cp_dooropen":4,"estimatedrange":"256","idealrange":"220","idealrange_max":"220","linevoltage":"0","m_msgage_d":84,"m_msgage_s":84,"m_msgtime_d":"2025-09-24 07:08:02","m_msgtime_s":"2025-09-24 07:08:02","mode":"range","pilotpresent":0,"soc":"100.0","soh":"84.98","staleambient":"1","staletemps":"1","temperature_ambient":"10","temperature_battery":"11.1094","temperature_cabin":"11","temperature_charger":"17","temperature_motor":"14.1719","temperature_pem":"18","units":"K","vehicle12v":"12.46","vehicle12v_current":"0","vehicle12v_ref":"12.54"}
[lp-2  ] DEBUG 2025/09/24 07:09:26 limitSoc reached: 100.0% > 95%
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.6 (16330685)

andig, 2025-09-24:
> instead to stop charging evcc try to continue charging.

And what is the problem?

mfuchs1984, 2025-09-24:
I suppose you get the warning since you mixed up charging and enabled. They are not the same, a charger can be enabled but the vehicle is not charging.

```
enabled:
    source: http
    uri: "http://ovms.dexters-web.de:6868/api/charge/WF-WM110E/?username=istler&password=jFiN%5EDg5V27%5EAQef33Je"
    jq: .chargestate == "charging" or .chargestate == "topoff" # parse response json
    
```

magicWF, 2025-09-24:
> > instead to stop charging evcc try to continue charging.
> 
> And what is the problem?

When I set the session target to 95% SoC and the vehicle SoC is greater (equal ?) than 95% SoC, than evcc not try to "enable" the wallbox. So I expected when the vehicle SoC is reached, that no more the enable statement was called.

Perhaps the problem is which @mfuchs1984 noticed, that I try to control the charging by the car. So I avoid the problem, that the car go to sleep in breaks during the PV charging.