evcc-io/evcc #23828: UI: problem reporting: Request failed with status code 404.

TobiasHuber1980, 2025-09-25:
## Description

Request failed with status code 404.
GET https://die.ip.adresse/api/config/evcc.yaml

Refs: https://github.com/evcc-io/evcc/pull/23707

## Steps to Reproduce

1) Klicke auf Hilfe benötigt
2) Klicke auf Bug melden (hier entsteht wohl schon der fehler) 

## Logs

```
[site  ] DEBUG 2025/09/25 08:43:15 battery 1 power: -1130W
[site  ] DEBUG 2025/09/25 08:43:15 grid power: 2W
[site  ] DEBUG 2025/09/25 08:43:15 !! solar production: accumulate add db:4 0.000W power (was: Accumulated: 2292.085kWh updated: 2025-09-25 08:43:01 +0200 CEST)
[site  ] DEBUG 2025/09/25 08:43:15 !! solar production: accumulate moved db:4 from 2292.085 to 2292.091
[site  ] DEBUG 2025/09/25 08:43:15 battery 1 soc: 12%
[site  ] DEBUG 2025/09/25 08:43:15 battery has priority at soc 12% (< 75%)
[site  ] DEBUG 2025/09/25 08:43:15 site power: 152W
[lp-1  ] DEBUG 2025/09/25 08:43:15 charge voltages: [231 233 232]V
[lp-1  ] DEBUG 2025/09/25 08:43:15 charge total import: 512.890kWh
[lp-1  ] DEBUG 2025/09/25 08:43:15 charger status: B
[lp-1  ] DEBUG 2025/09/25 08:43:15 pv charge current: 0A = 0A + -0.659A (152W @ 1p)
[site  ] DEBUG 2025/09/25 08:43:15 solar forecast: accumulated 0.003Wh from 2025-09-25 08:43:01 +0200 CEST to 2025-09-25 08:43:15 +0200 CEST
[site  ] DEBUG 2025/09/25 08:43:15 solar forecast: produced 2292.091
[site  ] DEBUG 2025/09/25 08:43:15 solar forecast: accumulated 3021.136kWh, produced 2292.091kWh, scale 0.759
[site  ] DEBUG 2025/09/25 08:43:30 ----
[lp-1  ] DEBUG 2025/09/25 08:43:30 charge power: 0W
[lp-1  ] DEBUG 2025/09/25 08:43:30 charge currents: [0 0 0]A
[site  ] DEBUG 2025/09/25 08:43:30 grid power: 2W
[site  ] DEBUG 2025/09/25 08:43:30 battery 1 power: -1130W
[site  ] DEBUG 2025/09/25 08:43:30 battery 1 soc: 12%
[site  ] DEBUG 2025/09/25 08:43:30 pv 1 power: 1431W
[site  ] DEBUG 2025/09/25 08:43:30 !! solar production: accumulate add db:4 0.000W power (was: Accumulated: 2292.091kWh updated: 2025-09-25 08:43:15 +0200 CEST)
[site  ] DEBUG 2025/09/25 08:43:30 !! solar production: accumulate moved db:4 from 2292.091 to 2292.097
[site  ] DEBUG 2025/09/25 08:43:30 battery has priority at soc 12% (< 75%)
[site  ] DEBUG 2025/09/25 08:43:30 site power: 152W
```

## Version

v0.207.6 (16330685)

andig, 2025-09-25:
Unter welcher Adresse läuft Dein UI?

spuky, 2025-09-25:
**die.ip.adresse** im cosoleoutput des browsers... ist eigentlich **der.reverse.proxy.hostname** wenn man schon ändert dann so dass der context bleibt ;-)

TobiasHuber1980, 2025-09-25:
ist auch lokal bei mir:
```
Request failed with status code 404.
GET http://192.168.1.34:7070/api/config/evcc.yaml
```

```
ailed to fetch config: 
ye
code
: 
"ERR_BAD_REQUEST"
config
: 
{transitional: {…}, adapter: Array(3), transformRequest: Array(1), transformResponse: Array(1), timeout: 0, …}
message
: 
"Request failed with status code 404"
name
: 
"AxiosError"
request
: 
XMLHttpRequest {onreadystatechange: null, readyState: 4, timeout: 0, withCredentials: false, upload: XMLHttpRequestUpload, …}
response
: 
{data: '{"error":"no config file found"}\n', status: 404, statusText: 'Not Found', headers: Vt, config: {…}, …}
status
: 
404
stack
: 
"AxiosError: Request failed with status code 404\n    at tb (http://192.168.1.34:7070/assets/index-DVkIo7JH.js:66:25610)\n    at XMLHttpRequest.m (http://192.168.1.34:7070/assets/index-DVkIo7JH.js:66:30231)\n    at Ws.request (http://192.168.1.34:7070/assets/index-DVkIo7JH.js:66:38733)\n    at async Proxy.loadYamlConfig (http://192.168.1.34:7070/assets/Issue-CsgC2YoF.js:1:11641)"
[[Prototype]]
: 
Error

```

spuky, 2025-09-25:
ok schaut auch bei mir so aus als ob der Fehler von evcc kommt...  weil der 404 Body so ausschaut:

`{"error":"no config file found"}`

spuky, 2025-09-25:
Ja im Log ist nichts... weshalb ich auch dachte ok wird der reverse proxy sein...und evcc sieht den request nicht... und nur die ui loggt ihn...

spuky, 2025-09-25:
http_config_yaml_handler.go L15
```
if configFilePath == "" {
			jsonError(w, http.StatusNotFound, fmt.Errorf("no config file found"))
			return
		}
```

der config file path schein empty zu sein...

spuky, 2025-09-25:
root.go handelt eine leere 'cfgFile' variable setzt die aber nie auf einen korrekten wert später in der httpd.RegisterSystemHandler wird wieder cfgFile verwendet...

```
func initConfig() {
	if cfgFile != "" {
		// Use config file from the flag
		viper.SetConfigFile(cfgFile)
	} else {
		// Search for config in home directory if available
		if home, err := os.UserHomeDir(); err == nil {
			viper.AddConfigPath(home)
		}

		// Search config in home directory with name "mbmd" (without extension).
		viper.AddConfigPath(".")    // optionally look for config in the working directory
		viper.AddConfigPath("/etc") // path to look for the config file in

		viper.SetConfigName("evcc")
	}
	if cfgDatabase != "" {
		viper.Set("Database.Dsn", cfgDatabase)
	}
}
```