evcc-io/evcc #23845: Polestar: panic: runtime error: index out of range [0] with length 0

mnggh, 2025-09-25:
### Describe the bug

Hi,
after waking up just to find that my Polestar 3 has not been charged during the night a couple of times, I started to investigate. It turned out, that evcc panicked when trying to access the Polestar API: "panic: runtime error: index out of range [0] with length 0".
After that evcc will be restarted because watchdog is active.

I then tried to simply use the guest vehicle to have my car charged but evcc would also restart. Interestingly, without panic.
I had somewhat better chance, when evcc was not able to connect to the Polestar API (strike-through cloud) at all.
No issue when charging my other car Smart #1.

### Steps to reproduce

1. Plug in car
2. Change mode from pv to fast
3. Wait
4. evcc will panic eventually

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error

# unique installation id
plant: <plant_id>

interval: 30s # control cycle interval

sponsortoken: <sponsortoken>

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- name: grid_huawei
  type: template
  template: huawei-sun2000
  usage: grid
  modbus: tcpip
  id: 1
  host: 10.10.70.17
  port: 502
  timeout: 15s # optional
- name: pv_huawei
  type: template
  template: huawei-sun2000
  usage: pv
  modbus: tcpip
  id: 1
  host: 10.10.70.17
  port: 502
  timeout: 15s # optional
- name: battery_huawei
  type: template
  template: huawei-sun2000
  usage: battery
  modbus: tcpip
  id: 1
  host: 10.10.70.17
  port: 502
  timeout: 15s # optional
- name: pv_sma
[... 262 lines cut ...]
            });
        }
    </script>
</body>
</html>
[polestar] TRACE 2025/09/24 22:41:41 POST https://polestarid.eu.polestar.com/as/<some_id>/resume/as/authorization.ping
[polestar] TRACE 2025/09/24 22:41:41 client_id=<client_id>pf.pass=***&pf.username=***
[polestar] TRACE 2025/09/24 22:41:41 GET https://www.polestar.com/sign-in-callback?code=<code_2>&state=<state_2>
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.6

andig, 2025-09-26:
Ein trace log wäre Klasse. Da scheinen Daten zu fehlen:

```go
// SOC via car telemetry
func (v *Provider) Soc() (float64, error) {
	res, err := v.telemetryG()
	return res.Battery[0].BatteryChargeLevelPercentage, err
}
```

mnggh, 2025-09-26:
Danke für die Antwort. Sehr gerne liefere ich einen trace log.
Wie in der Config oben und auch im Log oben zu sehen, steht der log level bereits auf trace.
Gibt es etwa noch eine andere Stelle, an der ich ein passende trace log ziehen kann?

andig, 2025-09-26:
Das crasht leider bevor der Aufruf im Log erscheint. Ich kann einen Workaround einbauen, es wäre super danach nochmal ein neues Log zu bekommen.

mnggh, 2025-09-26:
Stark, vielen Dank! Ich werde den Test am WE per nightly wiederholen und melde mich. Bis bald!

andig, 2025-09-27:
Fixed again in 84a74480cdd97c9be478d4bdbc8bace3922ac924. Still need new trace after that to see where the data went...

closed 2025-09-27 by andig

mnggh, 2025-09-28:
First of all: No panics anymore. Thanks!
However, there's still something odd here, as sometimes there's a bunch of gibberish in the logs. Please find a few log snippets below.

10:14:43 (broken)
```
[mqtt  ] TRACE 2025/09/28 10:14:43 send evcc/loadpoints/1/sessionPrice: ''
[cache ] TRACE 2025/09/28 10:14:43 lp-1/chargedEnergy: 0
[polestar] TRACE 2025/09/28 10:14:43 POST https://pc-api.polestar.com/eu-north-1/mystar-v2
[mqtt  ] TRACE 2025/09/28 10:14:43 send evcc/loadpoints/1/chargedEnergy: '0'
[mqtt  ] TRACE 2025/09/28 10:14:43 send evcc/loadpoints/1/sessionCo2PerKWh: ''
[cache ] TRACE 2025/09/28 10:14:43 lp-1/offeredCurrent: 0
[mqtt  ] TRACE 2025/09/28 10:14:43 send evcc/loadpoints/1/sessionPricePerKWh: ''
[cache ] TRACE 2025/09/28 10:14:43 lp-1/vehicleWelcomeActive: false
[mqtt  ] TRACE 2025/09/28 10:14:43 send evcc/loadpoints/1/vehicleWelcomeActive: 'false'
[mqtt  ] TRACE 2025/09/28 10:14:43 send evcc/loadpoints/1/offeredCurrent: '0'
[mqtt  ] TRACE 2025/09/28 10:14:43 send evcc/loadpoints/1/connected: 'true'
[cache ] TRACE 2025/09/28 10:14:43 lp-1/connected: true
[mqtt  ] TRACE 2025/09/28 10:14:43 send evcc/loadpoints/1/charging: 'false'
[cache ] TRACE 2025/09/28 10:14:43 lp-1/charging: false
[proxy-5021] TRACE 2025/09/28 10:14:43 read holding: 00 06 00 02 00 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2e 89 0a 7a 00 90 0d e0 00 5b
[proxy-5021] TRACE 2025/09/28 10:14:43 read holdings: id 1 addr 32064 qty 52
[proxy-5021] TRACE 2025/09/28 10:14:44 read holding: 00 00 02 96 0f bf 0f b6 0f cc 09 19 09 12 09 1c 00 00 01 77 00 00 01 79 00 00 01 89 00 00 02 79 00 00 01 00 ff ff ff fd 03 e8 13 89 27 10 01 c6 0b b8 02 00 00 00 68 d8 e4 72 68 d8 47 84 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 01 50 ea 00 14 9a 19 00 16 ae bf 68 d9 0a 8f 00 00 00 0a 00 00 00 43
[polestar] TRACE 2025/09/28 10:14:44 {"query":"query CarTelematicsV2($vins:[String!]!){carTelematicsV2(vins: $vins){health{vin,brakeFluidLevelWarning,daysToService,distanceToServiceKm,engineCoolantLevelWarning,oilLevelWarning,serviceWarning,timestamp{seconds,nanos}},battery{vin,batteryChargeLevelPercentage,chargingStatus,estimatedChargingTimeToFullMinutes,estimatedDistanceToEmptyKm,timestamp{seconds,nanos}},odometer{vin,odometerMeters,timestamp{seconds,nanos}}}}","variables":{"vins":["***"]},"operationName":"CarTelematicsV2"}
--
�      ��Mk�@��˜s�Mj>�f5���Ҥ�i2���F�Ad�{	����=���
Zdu���Yd���x<�
�@��`�>�Od��
jw��v�`[m��rZ��Q�I���K�N�iM2��7��*)"h�J�c�ȃ����b��8T��~�����ȳ���ξjm��c6�
L���͵gt
�]i�|]YPr""`m�3��8�Sӹv��lR��,�
����<��,��4�y�������Y��m���B��i*��R�,�"	aB�  �� J�x�  
[main  ] DEBUG 2025/09/28 10:14:44 vehicle status: A (P-3CO)
[alfen ] TRACE 2025/09/28 10:14:44 modbus: send 00 1a 00 00 00 06 01 03 04 ba 00 02
[alfen ] TRACE 2025/09/28 10:14:44 modbus: recv 00 1a 00 00 00 07 01 03 04 00 00 00 00
[lp-1  ] DEBUG 2025/09/28 10:14:44 available power -100W < 4140W min 3p threshold
[cache ] TRACE 2025/09/28 10:14:44 lp-1/enabled: false
[mqtt  ] TRACE 2025/09/28 10:14:44 send evcc/updated: '1759047284'
[cache ] TRACE 2025/09/28 10:14:44 lp-1/mode: pv
[mqtt  ] TRACE 2025/09/28 10:14:44 send evcc/loadpoints/1/planOverrun: '0'
```

10:30:16 (broken)
```
[alfen ] TRACE 2025/09/28 10:30:16 modbus: recv 01 3d 00 00 00 0d 01 03 0a 43 32 00 00 00 00 00 00 00 00
[lp-1  ] DEBUG 2025/09/28 10:30:16 charger status: C
[mqtt  ] TRACE 2025/09/28 10:30:16 send evcc/loadpoints/1/connected: 'true'
[polestar] TRACE 2025/09/28 10:30:16 POST https://polestarid.eu.polestar.com/as/token.oauth2
[cache ] TRACE 2025/09/28 10:30:16 lp-1/vehicleWelcomeActive: false
[mqtt  ] TRACE 2025/09/28 10:30:16 send evcc/loadpoints/1/vehicleWelcomeActive: 'false'
[cache ] TRACE 2025/09/28 10:30:16 lp-1/connected: true
[cache ] TRACE 2025/09/28 10:30:16 lp-1/charging: true
[mqtt  ] TRACE 2025/09/28 10:30:16 send evcc/loadpoints/1/charging: 'true'
[polestar] TRACE 2025/09/28 10:30:16 client_id=<client_id>&grant_type=refresh_token&refresh_token=<refresh_token>
--
{"access_token":"<access_token>","refresh_token":"<refresh_token>","token_type":"Bearer","expires_in":299}
[polestar] TRACE 2025/09/28 10:30:16 POST https://pc-api.polestar.com/eu-north-1/mystar-v2
[proxy-5021] TRACE 2025/09/28 10:30:16 read holding: 00 00 02 26 0f ab 0f 9a 0f b2 09 14 08 ff 09 0b 00 00 03 1a 00 00 03 20 00 00 03 2f 00 00 02 79 00 00 02 27 ff ff ff ff 03 e8 13 85 27 10 01 c9 0b b8 02 00 00 00 68 d8 e4 72 68 d8 47 84 00 00 02 26 00 00 00 00 00 00 00 00 00 00 ff ff ff b9 00 01 50 ea 00 14 9a 27 00 16 ae d0 68 d9 0e 29 00 00 00 18 00 00 00 51
[proxy-5021] TRACE 2025/09/28 10:30:16 read holdings: id 1 addr 37100 qty 38
[proxy-5021] TRACE 2025/09/28 10:30:17 read holding: 00 01 00 00 09 0e 00 00 08 f8 00 00 09 04 ff ff f9 d7 ff ff f9 d1 ff ff fa 02 ff ff d6 17 00 00 03 f8 03 de 13 85 00 0e a9 90 00 11 02 6e 00 00 00 00 00 01 00 00 0f 9b 00 00 0f 93 00 00 0f a7 ff ff f2 0d ff ff f1 da ff ff f2 30
[proxy-5021] TRACE 2025/09/28 10:30:17 read holdings: id 1 addr 37760 qty 28
[polestar] TRACE 2025/09/28 10:30:18 {"query":"query CarTelematicsV2($vins:[String!]!){carTelematicsV2(vins: $vins){health{vin,brakeFluidLevelWarning,daysToService,distanceToServiceKm,engineCoolantLevelWarning,oilLevelWarning,serviceWarning,timestamp{seconds,nanos}},battery{vin,batteryChargeLevelPercentage,chargingStatus,estimatedChargingTimeToFullMinutes,estimatedDistanceToEmptyKm,timestamp{seconds,nanos}},odometer{vin,odometerMeters,timestamp{seconds,nanos}}}}","variables":{"vins [... cut ...]
[alfen ] TRACE 2025/09/28 10:45:20 modbus: send 02 54 00 00 00 06 01 03 04 ba 00 02
[mqtt  ] TRACE 2025/09/28 10:45:20 send evcc/loadpoints/1/chargeRemainingDuration: '25432'
[cache ] TRACE 2025/09/28 10:45:20 lp-1/vehicleSoc: 43.32149532710299
[cache ] TRACE 2025/09/28 10:45:20 lp-1/chargeRemainingDuration: 7h3m52s
[cache ] TRACE 2025/09/28 10:45:20 lp-1/chargeRemainingEnergy: 67384.44444444422
[mqtt  ] TRACE 2025/09/28 10:45:20 send evcc/updated: '1759049120'
[mqtt  ] TRACE 2025/09/28 10:45:20 send evcc/loadpoints/1/vehicleSoc: '43.321'
```

Hope this helps.

andig, 2025-09-28:
Unfortuantely not. Seems the response is gzip'ed, need to look into decoding this: https://github.com/hasura/go-graphql-client/issues/181.