evcc-io/evcc #23867: Evcc lässt sich nicht aktualisieren

skipperkm, 2025-09-27:
### Describe the bug

Raspi 4 
nach update und Upgrade bleibt evcc auf Version 0.207.6 hängen.
Wird als aktuellste Version angegeben

### Steps to reproduce

apt Update 
Usw

### Configuration details

```yaml
na
```

### Log details

```text
na
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.6

VolkerK62, 2025-09-27:
>I have verified that the issue is reproducible with the latest nightly build

das aktuellen Nightly funktioniert
`evcc version 0.208.0 (4a4027c7)`

Maschga, 2025-09-27:
Bei der `0.208.0` wird Immutable angezeigt:
[image]

Das ist bei den anderen Releases nicht der Fall, hier z.B. die `0.207.6`:

[image]

Im Log steht mehrmals diese Fehlermeldung: `422 Cannot upload assets to an immutable release.`

Wahrscheinlich ist #23796 das Problem.
Ich gehe davon aus, dass das Ändern von `read` auf `write` in der `release.yml` den Fehler beheben würde. (See [Github Docs](https://docs.github.com/en/actions/reference/workflows-and-actions/workflow-syntax#permissions): `write` includes `read`)

Brezentrager, 2025-09-27:
0.208.1 works for me. 

(0.208.0 didn't)