evcc-io/evcc #23868: Different tariffs per vehicle / driver

64ralf, 2025-09-27:
**Kurzer Hinweis (Deutsch):**  
Ich nutze evcc mit mehreren Fahrzeugen und Fahrern und möchte die Kosten pro Fahrzeug/Fahrer getrennt abrechnen.  
Dafür wäre es hilfreich, wenn evcc unterschiedliche Tarife (€/kWh) pro Fahrzeug oder Fahrer unterstützen würde.  
Die Details dazu habe ich unten auf Englisch formuliert.

# 🚀 Feature Request: Different tariffs per vehicle / driver

### Is your feature request related to a problem? Please describe.
Currently, tariffs (€/kWh) in evcc can only be configured **globally**.  
When multiple vehicles and drivers share the same charging infrastructure, it is not possible to account for charging costs separately per vehicle or driver.

**Example use case:**  
- Driver A should be billed at €0.30/kWh.  
- Driver B should be billed at €0.25/kWh.  
- In the charging session statistics, costs should be calculated accordingly.  

At the moment, this requires exporting session data and doing manual post-processing (e.g., Excel).

---

### Describe the solution you'd like
- Ability to configure a **tariff per vehicle** (fixed or dynamic).  
- Alternatively: assign **tariffs per driver** (via vehicle assignment or tags).  
- Charging sessions and statistics should then reflect the costs based on the assigned tariff.

---

### Describe alternatives you’ve considered
- Running multiple evcc instances (impractical).  
- Using time-based tariff zones and assigning drivers fixed charging times (very inflexible).  
- Manual export and external calculation (time-consuming and error-prone).  

---

### Additional context
This feature would simplify cost allocation for households or workplaces with multiple vehicles and drivers, and increase flexibility for shared charging infrastructure.

andig, 2025-09-27:
@64ralf wieviele Fahrer/ Boxen hast du denn dass das ein Thema ist?

64ralf, 2025-09-27:
@andig   Ich habe ein Objekt mit 2 Ladepunkten und aktuell 6 unterschiedlichen Fahrzeugen die sich mit RFID anmelden. Es kommen aber noch 4 weiter Fahrzeuge dazu. Es soll der Bezug über den Ladepunkt (Wallbox) verrechnet werden und dies zu teilweise unterschiedlicher Kosten/kWh per Fahrzeug.

I have a property with 2 charging points and currently 6 different vehicles that log in using RFID. However, 4 more vehicles will be added. The charging should be billed via the charging point (wallbox), with partially different costs per kWh depending on the vehicle.

naltatis, 2025-09-28:
Die Kosten, die wir pro Ladevorgang errechnen und speichern sind ja eine Mischung aus entgangener Einspeisevergütung und Netzbezugspreis, also abhängig vom Sonnenanteil. Wäre dein Ziel diese Mechanik weiter zu erhalten oder möchtest du, dass für die Ladesession dann auch der Anteil Eigenenergie ignoriert wird?

64ralf, 2025-09-28:
@naltatis In diesem Fall kann der Anteil der Eigenenergie ignoriert werden. Es Gght um die reine Verrechnung von bezogenen kWh der unterschiedlichen Fahrzeugen. Die Kosten/kWh wurde vorher mit den Nutzern vereinbart und sind fix.

naltatis, 2025-09-28:
Ok, das heißt dass das Post-Processing via Excel ist ein reines Fahrzeug-Energiemenge x Fahrzeugpreis.