evcc-io/evcc #23869: Nach Update auf 0.208 keine Zendure mehr Verfügbar bzw bringt Fehler

Nekromonger, 2025-09-27:
### Describe the bug

Nach Update auf 0.208 kommt immer in der Oberfläche der Fehler

cannot create meter type 'template': cannot create meter type 'zendure': decoding failed due to the following error(s): '' has invalid keys: capacity

vorher ging noch alles

Fehler tritt sowohl im Docker Container als auch im homeassistant addon auf

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# replace with your real pv system
#  - name: my_pv
#    type: template
#    template: zendure
#    usage: pv
#    account: 
#    serial:  # Zu finden in der Zendure App in den Einstellungen des Geräts
#    region: GLOBAL # optional
#    timeout: 10s # optional
  # replace with your real battery
#  - name: my_battery
#    type: template
#    template: zendure
#    usage: battery
#    account: 
#    serial:  # Zu finden in der Zendure App in den Einstellungen des Geräts
#    region: GLOBAL # optional
#    timeout: 10s # optional
```

### Log details

```text
[main  ] FATAL 2025/09/27 10:04:07 meter [my_pv] cannot create meter 'my_pv': cannot create meter type 'template': cannot create meter type 'zendure': decoding failed due to the following error(s):
'' has invalid keys: capacity
failed booting site: not found: my_pv
[main  ] FATAL 2025/09/27 10:04:07 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-09-27:
PV kann keine Kapazität haben.

andig, 2025-09-27:
@premultiply Ursache ist https://github.com/evcc-io/evcc/pull/23776 in Verbindung mit einem Defaultwert. Damit ist PV kaputt.

closed 2025-09-27 by andig