evcc-io/evcc #23873: Laden startet trotz priorisierter Hausbatterie

michiproep, 2025-09-27:
### Describe the bug

Hallo,
ich hatte heute ein sehr komisches Verhalten, das ich mir nicht erklären kann und auch zuvor noch nie beobachtet habe.
Zum Zeitpunkt hatte ich folgende Batterie-Einstellung: (eigentlich sollte hier kein Laden starten)
Und alle Ladepunkte stehen auf PV-Modus und die minimalen Socs sind nirgends unterschritten.
[image]

Jetzt habe das Auto eingesteck und es hat sofort zum Laden angefangen.
Nach ca. 5 Minuten hat sich dann der nächste Ladepunkt ebenfalls zugeschalten - trotz Batterieeinstellung.
Nach weiteren 3 Minuten (was ja ok ist) wurde dann das Auto wieder abgeschalten und auch der zweite Ladepunkt hat den OffTimer gestartet

[image]

Bin mir nicht sicher ob ich noch irgendwas übersehen habe, aber eigentlich sollte das Laden nicht starten, oder?

Leider ist das Logfile etwas zu spät, aber ich häng's trotzdem mal dazu

[evcc-20250927-114728-debug.log](https://github.com/user-attachments/files/22572417/evcc-20250927-114728-debug.log)

---
I have verified that the issue is reproducible with the latest nightly build* => no

### Steps to reproduce

1. Batterieeinstellung: Priorisiere Batterie > aktueller Soc
2. PV modus + minSoc Einstellung zB 1%
3. Auto anstecken
...

### Configuration details

```yaml
vehicles:
- name: bmwix2
  type: template
  template: ioBroker.bmw
  title: iX2
  capacity: 66 # Akkukapazität in kWh (optional)
  vin: XXXX# Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
  uri: http://iobroker:8082
  icon: car # Icon in der Benutzeroberfläche (optional)
  # welcomecharge: false
```

### Log details

```text
[site  ] DEBUG 2025/09/27 11:45:45 pv 2 power: 4387W
[site  ] DEBUG 2025/09/27 11:45:45 battery 1 power: -10752W
[site  ] DEBUG 2025/09/27 11:45:45 pv 1 power: 10141W
[site  ] DEBUG 2025/09/27 11:45:45 battery 1 soc: 41%
[site  ] DEBUG 2025/09/27 11:45:45 pv power: 14528W
[site  ] DEBUG 2025/09/27 11:45:45 !! solar production: accumulate set stpse10 22508.966kWh meter total (was: Accumulated: 6239.485kWh updated: 2025-09-27 11:45:39 +0200 CEST meter: 22508.966kWh)
[site  ] DEBUG 2025/09/27 11:45:45 !! solar production: accumulate moved stpse10 from 6239.485 to 6239.485
[site  ] DEBUG 2025/09/27 11:45:45 !! solar production: accumulate set stp6 11545.979kWh meter total (was: Accumulated: 3254.330kWh updated: 2025-09-27 11:45:39 +0200 CEST meter: 11545.972kWh)
[site  ] DEBUG 2025/09/27 11:45:45 !! solar production: accumulate moved stp6 from 3254.330 to 3254.337
[site  ] DEBUG 2025/09/27 11:45:45 battery has priority at soc 41% (< 53%)
[site  ] DEBUG 2025/09/27 11:45:45 site power: -322W
[lp-1  ] DEBUG 2025/09/27 11:45:45 charge voltages: [237 238 239]V
[lp-1  ] DEBUG 2025/09/27 11:45:45 detected connected phases: 3p
[lp-1  ] DEBUG 2025/09/27 11:45:45 charge total import: 7933.926kWh
[lp-1  ] DEBUG 2025/09/27 11:45:45 charger status: B
[lp-1  ] DEBUG 2025/09/27 11:45:45 next soc poll remaining time: 15s
[lp-1  ] DEBUG 2025/09/27 11:45:45 pv charge current: 0.467A = 0A + 0.467A (-322W @ 3p)
[site  ] DEBUG 2025/09/27 11:45:45 solar forecast: accumulated 0.007Wh from 2025-09-27 11:45:39 +0200 CEST to 2025-09-27 11:45:45 +0200 CEST
[site  ] DEBUG 2025/09/27 11:45:45 solar forecast: produced 9493.822
[site  ] DEBUG 2025/09/27 11:45:45 solar forecast: accumulated 9118.167kWh, produced 9493.822kWh, scale 1.041
[lp-3  ] DEBUG 2025/09/27 11:45:50 set session soc limit: 36
[site  ] DEBUG 2025/09/27 11:45:51 ----
[... 488 lines cut ...]
[site  ] DEBUG 2025/09/27 11:47:27 !! solar production: accumulate moved stp6 from 3254.420 to 3254.424
[site  ] DEBUG 2025/09/27 11:47:27 battery 1 soc: 43%
[site  ] DEBUG 2025/09/27 11:47:27 battery has priority at soc 43% (< 53%)
[site  ] DEBUG 2025/09/27 11:47:27 site power: 31W
[lp-3  ] DEBUG 2025/09/27 11:47:27 charger status: A
[site  ] DEBUG 2025/09/27 11:47:27 solar forecast: accumulated 0.007Wh from 2025-09-27 11:47:21 +0200 CEST to 2025-09-27 11:47:27 +0200 CEST
[site  ] DEBUG 2025/09/27 11:47:27 solar forecast: produced 9493.915
[site  ] DEBUG 2025/09/27 11:47:27 solar forecast: accumulated 9118.290kWh, produced 9493.915kWh, scale 1.041
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.5

michiproep, 2025-09-27:
Ja, ich weiß, dass das fehlt. Aber das bekomm ich leider nicht rückwirkend. Zumindest wüsste ich nicht wie...
Alles was ich sagen kann, ist: Eingesteckt und es ging sofort los...

Richtig, LP2 hat höhere Prio. 
Das erklärt vielleicht warum er startet: Wenn LP1 aktiv und LP2 höhere Prio, dann Lp1 aus und Lp2 an...? weiß nicht wie evcc das intern regelt?
Sollte aber auch nicht so sein, wenn die Batterie unterhalb des erlaubten Levels ist.

VolkerK62, 2025-09-27:
Bzgl Log, evtl hilft das weiter https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen  Am Ende ist auch ein Link bzgl Docker.
>Richtig, LP2 hat höhere Prio.
Das erklärt vielleicht warum er startet: Wenn LP1 aktiv und LP2 höhere Prio, dann Lp1 aus und Lp2 an...? weiß nicht wie evcc das intern regelt?

Genau so. Die Ladeleistung von LP1 wird LP2 zur Verfügung gestellt.

>Sollte aber auch nicht so sein, wenn die Batterie unterhalb des erlaubten Levels ist.

Ist ja nur ein Folgefehler, da LP1 gar nicht laden dürfte.

andig, 2025-10-04:
Ohne ins Log zu schauen: Batterieprio heisst *nur*, dass die Batterieladeleistung *nicht* als fürs Fahrzeug verfügbar angesehen wird. Wenn ein Fahrzeug aber mal lädt (und deshalb die Batterie nicht) dann hört es damit im statischen Zustand auch nicht wieder auf (es gibt ja keine Batterieladeleistung aufgrund derer es seinen Anteil reduzieren müsste). Normalerweise ist das kein Problem da die Batteriesteuerung immer schneller ist als evcc und sich daher nach und nach jede kleinste verfügbare Leistung schnappen würde. Jetzt müsstest du selbst nochmal im Log prüfen, ob das hier der Fall sein kann.

/cc @VolkerK62 @premultiply

michiproep, 2025-10-04:
Hi @andig , danke für die Info. 
"...wenn ein Fahrzeug aber mal lädt..." Den Teil verstehe ich, aber
`
[site  ] DEBUG 2025/10/04 12:06:38 battery has priority at soc 52% (< 56%)
[lp-5  ] DEBUG 2025/10/04 12:06:38 charger status: B
[lp-5  ] DEBUG 2025/10/04 12:06:38 pv enable in 30s
[lp-5  ] DEBUG 2025/10/04 12:06:38 pv enable timer remaining: 30s
`