evcc-io/evcc #23880: Fronius Gen 24 - Fehler im Log seit Update auf 0.208

papaloewe, 2025-09-27:
### Describe the bug

Nach Update auf 0.208 und auch 0.208.1 erscheint ein Fehler im LOG.

### Steps to reproduce

1.Update EVCC auf aktuelle Version
2.Starte EVCC mit der Konfiguration eines Fronius Gen24 Geräts
3.
...

### Configuration details

```yaml
Konfig:

`meters:
    - name: pv2
    type: template
    template: fronius-gen24
    usage: pv
    host: 192.168.30.3
    port: 502
    maxacpower: 8000 # Maximale AC Leistung des Hybrid-Wechselrichters (optional)
    maxchargerate: 100 # Leistungslimit für Netzladung von PV in %. (def. 100%, optional)`
```

### Log details

```text
Log:
`2025-09-27 15:55:03.094386542 +0200 CEST m=+545.478414416 !!   0ms invalid character 'u' looking for beginning of value
[site  ] ERROR 2025/09/27 15:55:03 pv 2 power: invalid character 'u' looking for beginning of value`
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.208.1

andig, 2025-09-27:
Bitte trace log

papaloewe, 2025-09-27:
```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/config/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/09/27 18:09:55 evcc 0.208.1
[main  ] INFO 2025/09/27 18:09:55 using config file: /config/evcc.yaml
[site  ] INFO 2025/09/27 18:10:00 site config:
[site  ] INFO 2025/09/27 18:10:00   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/09/27 18:10:00     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/09/27 18:10:00     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/09/27 18:10:00     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/09/27 18:10:00     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2025/09/27 18:10:00   vehicles:
[site  ] INFO 2025/09/27 18:10:00     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/09/27 18:10:00   tariffs:
[site  ] INFO 2025/09/27 18:10:00     grid:      priceforecast
[site  ] INFO 2025/09/27 18:10:00     feed-in:   pricestatic
[site  ] INFO 2025/09/27 18:10:00     co2:       ✓
[site  ] INFO 2025/09/27 18:10:00     solar:     ✓
[site  ] DEBUG 2025/09/27 18:10:00 set buffer soc: 100
[site  ] DEBUG 2025/09/27 18:10:00 set priority soc: 100
[site  ] DEBUG 2025/09/27 18:10:00 set battery discharge control: true
[site  ] DEBUG 2025/09/27 18:10:00 accumulated solar yield: restored 2585.514kWh forecasted, map[pv2:{clock:<nil> updated:{wall:0 ext:0 loc:<nil>} meter:<nil> Accumulated:2078.1925200000005} pv_no1:{clock:<nil> updated:{wall:0 ext:0 loc:<nil>} meter:<nil> Accumulated:196.66918445633132}] produced
[site  ] DEBUG 2025/09/27 18:10:00 solar forecast: accumulated 1.995Wh from 0001-01-01 00:00:00 +0000 UTC to 2025-09-27 18:10:00 +0200 CEST
[site  ] DEBUG 2025/09/27 18:10:00 solar forecast: produced 2274.862
[site  ] DEBUG 2025/09/27 18:10:00 solar forecast: accumulated 2585.514kWh, produced 2274.862kWh, scale 0.880
[site  ] DEBUG 2025/09/27 18:10:04 ----
[site  ] ERROR 2025/09/27 18:10:16 
2025-09-27 18:10:16.593118412 +0200 CEST m=+20.713110172 !!   0ms invalid character 'u' looking for beginning of value
[site  ] ERROR 2025/09/27 18:10:16 pv 2 power: invalid character 'u' looking for beginning of value
[site  ] DEBUG 2025/09/27 18:10:16 grid power: 1W
[site  ] DEBUG 2025/09/27 18:10:16 battery 1 power: 1783W
[site  ] DEBUG 2025/09/27 18:10:16 grid powers: [961 -366 -593]W
[site  ] DEBUG 2025/09/27 18:10:16 pv 1 power: 239W
[site  ] DEBUG 2025/09/27 18:10:16 battery 1 soc: 67%
[site  ] DEBUG 2025/09/27 18:10:16 grid currents: [4.2 -2.04 -2.6]A
[site  ] DEBUG 2025/09/27 18:10:16 pv power: 239W
[site  ] DEBUG 2025/09/27 18:10:16 !! solar production: accumulate set pv2 10792.774kWh meter total (was: Accumulated: 2078.193kWh updated: 0001-01-01 00:00:00 +0000 UTC)
[site  ] DEBUG 2025/09/27 18:10:16 !! solar production: accumulate moved pv2 from 2078.193 to 2078.193
[site  ] DEBUG 2025/09/27 18:10:16 !! solar production: accumulate add pv_no1 0.000W power (was: Accumulated: 196.669kWh updated: 0001-01-01 00:00:00 +0000 UTC)
[site  ] DEBUG 2025/09/27 18:10:16 !! solar production: accumulate moved pv_no1 from 196.669 to 196.669
[site  ] DEBUG 2025/09/27 18:10:16 site power: 1984W
[site  ] DEBUG 2025/09/27 18:10:41 solar forecast: accumulated 0.013Wh from 2025-09-27 18:10:00 +0200 CEST to 2025-09-27 18:10:41 +0200 CEST
[site  ] DEBUG 2025/09/27 18:10:41 solar forecast: produced 2274.862
[site  ] DEBUG 2025/09/27 18:10:41 solar forecast: accumulated 2585.526kWh, produced 2274.862kWh, scale 0.880

andig, 2025-09-27:
https://github.com/evcc-io/evcc/issues/23880#issuecomment-3341867339

papaloewe, 2025-10-02:
[evcc-20251002-182835-trace.log](https://github.com/user-attachments/files/22663841/evcc-20251002-182835-trace.log)