evcc-io/evcc #23888: Nach Update auf 0.208.1: SMA SHM 2.0 zeigt keine Daten mehr

Klempner22, 2025-09-27:
### Describe the bug

Hallo Zusammen,

vielleicht hat jemand noch eine Idee was ich ausprobieren könnte um wieder die Daten von meiner Wallbox im Sunny Home Portal zu empfangen:

- evcc arbeiten aus meiner Wahrnehmung ansonsten fehlerfrei
- Ich habe gestartet mit einer geänderten yamel, in der ich die vendorid und die deviceid benutzt habe
  die Darstellung im Sunny Portal war ok, aber wurde es gemeldet keine Daten empfangen, Status unterbrochen
- Neustart SHM 2.0
- keine Änderungen
- Das selbe habe ich nochmal ohne die vendorid und die deviceid durchgeführt ==> keine Änderung
- Danach habe ich im Sunny Portal die "Wallbox-evcc" gelöscht und neu angelegt
- Der Verbraucher wird direkt gefunden und hat auch die alten Daten (SMA Seriennummer) und wird auch entsprechend erkannt, aber nicht mehr als Wallbox dargestellt, siehe Screenshoots.

Die ausgelesenen Parameter sehen für mich auch ok aus.

Hat jemand eine Idee, was ich falsch mache ? Oder handelt es sich um einen Bug ?

Ich habe mal noch einen Zeitschritt aus dem log angefügt, aber wie gesagt die Funktion von evcc ist ok,

[image]

[image]

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
hems:

# Neu ab V0.208.0
shm:
  allowcontrol: false # keine Steuerung ueber das Sunny-Portal
 # vendorid: "28081973" # Ausgelesen mit 192.xxx:7070/semp/
 # deviceid: "c642b49f7b0e" #
```

### Log details

```text
[site ] DEBUG 2025/09/27 19:02:28 ----
[lp-1 ] DEBUG 2025/09/27 19:02:28 charge power: 0W
[lp-1 ] DEBUG 2025/09/27 19:02:28 charge currents: [0 0 0]A
[site ] DEBUG 2025/09/27 19:02:28 grid power: 17W
[site ] DEBUG 2025/09/27 19:02:28 grid powers: [1399 -672 -710]W
[site ] DEBUG 2025/09/27 19:02:28 grid currents: [6.06 -2.96 -3.15]A
[site ] DEBUG 2025/09/27 19:02:28 pv 2 power: 0W
[site ] DEBUG 2025/09/27 19:02:28 battery 1 power: 2794W
[site ] DEBUG 2025/09/27 19:02:28 pv 1 power: 0W
[site ] DEBUG 2025/09/27 19:02:28 battery 1 soc: 18%
[site ] DEBUG 2025/09/27 19:02:28 pv power: 0W
[site ] DEBUG 2025/09/27 19:02:28 !! solar production: accumulate set pv1 12646.261kWh meter total (was: Accumulated: 540.239kWh updated: 2025-09-27 19:01:59 +0200 CEST meter: 12646.250kWh)
[site ] DEBUG 2025/09/27 19:02:28 !! solar production: accumulate moved pv1 from 540.239 to 540.250
[site ] DEBUG 2025/09/27 19:02:28 !! solar production: accumulate set pv_balkon 26.819kWh meter total (was: Accumulated: 52.880kWh updated: 2025-09-27 19:01:59 +0200 CEST meter: 26.819kWh)
[site ] DEBUG 2025/09/27 19:02:28 !! solar production: accumulate moved pv_balkon from 52.880 to 52.880
[site ] DEBUG 2025/09/27 19:02:28 site power: 3011W
[lp-1 ] DEBUG 2025/09/27 19:02:28 charge voltages: [235 238 237]V
[lp-1 ] DEBUG 2025/09/27 19:02:29 charge total import: 1942.335kWh
[lp-1 ] DEBUG 2025/09/27 19:02:29 charger status: A
[lp-1 ] DEBUG 2025/09/27 19:02:29 next soc poll remaining time: 5h36m28s
[site ] DEBUG 2025/09/27 19:02:29 solar forecast: accumulated 0.001Wh from 2025-09-27 19:01:59 +0200 CEST to 2025-09-27 19:02:29 +0200 CEST
[site ] DEBUG 2025/09/27 19:02:29 solar forecast: produced 593.130
[site ] DEBUG 2025/09/27 19:02:29 solar forecast: accumulated 677.718kWh, produced 593.130kWh, scale 0.875
[site ] DEBUG 2025/09/27 19:02:58 ----
[lp-1 ] DEBUG 2025/09/27 19:02:58 charge power: 0W
[lp-1 ] DEBUG 2025/09/27 19:02:58 charge currents: [0 0 0]A
[site ] DEBUG 2025/09/27 19:02:58 grid power: -15W
[site ] DEBUG 2025/09/27 19:02:58 grid powers: [1357 -656 -716]W
[site ] DEBUG 2025/09/27 19:02:58 grid currents: [5.86 -2.89 -3.17]A
[site ] DEBUG 2025/09/27 19:02:58 pv 2 power: 0W
[site ] DEBUG 2025/09/27 19:02:58 battery 1 power: 2805W
[site ] DEBUG 2025/09/27 19:02:58 pv 1 power: 0W
[site ] DEBUG 2025/09/27 19:02:58 pv power: 0W
[site ] DEBUG 2025/09/27 19:02:58 !! solar production: accumulate set pv1 12646.285kWh meter total (was: Accumulated: 540.250kWh updated: 2025-09-27 19:02:28 +0200 CEST meter: 12646.261kWh)
[site ] DEBUG 2025/09/27 19:02:58 !! solar production: accumulate moved pv1 from 540.250 to 540.274
[site ] DEBUG 2025/09/27 19:02:58 !! solar production: accumulate set pv_balkon 26.819kWh meter total (was: Accumulated: 52.880kWh updated: 2025-09-27 19:02:28 +0200 CEST meter: 26.819kWh)
[site ] DEBUG 2025/09/27 19:02:58 !! solar production: accumulate moved pv_balkon from 52.880 to 52.880
[site ] DEBUG 2025/09/27 19:02:59 battery 1 soc: 18%
[site ] DEBUG 2025/09/27 19:02:59 site power: 2990W
[lp-1 ] DEBUG 2025/09/27 19:02:59 charge voltages: [235 237 237]V
[lp-1 ] DEBUG 2025/09/27 19:02:59 charge total import: 1942.335kWh
[lp-1 ] DEBUG 2025/09/27 19:02:59 charger status: A
[lp-1 ] DEBUG 2025/09/27 19:02:59 next soc poll remaining time: 5h35m58s
[site ] DEBUG 2025/09/27 19:02:59 solar forecast: accumulated 0.001Wh from 2025-09-27 19:02:29 +0200 CEST to 2025-09-27 19:02:59 +0200 CEST
[site ] DEBUG 2025/09/27 19:02:59 solar forecast: produced 593.154
[site ] DEBUG 2025/09/27 19:02:59 solar forecast: accumulated 677.718kWh, produced 593.154kWh, scale 0.875
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.208.1

StefanSchoof, 2025-09-27:
Die Breaking Changes aus den Release notes hast du beacht? https://github.com/evcc-io/evcc/releases/tag/0.208.0

Klempner22, 2025-09-27:
Könnte mein Problem auch mit dem in #23889 beschrieben Problem der Gross- und Kleinschreibung zusammen hängen ?

In der Version: 0.207.6 habe ich folgendes ausgelesen:   **<DeviceId>F-28081973-c642b49f7b0e-00</DeviceId>**

in V0.208.1 steht:  <DeviceId>**F-28081973-C642B49F7B0E-00**</DeviceId>

Kann es sein das das Sunny Portal nicht in allen Bereichen nach Gross- und Kleinschreibung unterscheidet, da mein Gerät nicht als zusätzliches Gerät eingefügt wird, sondern immer meine früher genutzte Beschreibung verwendet, aber keine Datenverbindung aufbaut ?

hybe2022, 2025-09-27:
Wenn sie nicht mehr erkannt wird, dann ist es ein Duplikat. Aber du hast folgende Parameter aus dem Portal gepostet 

Typenschild	GerÃ¤te-ID	F-28081973-C642B49F7B0E-00

Da ist die ID komplett Capital. Sonst hätte ich dich schon zuvor darauf hingewiesen. Somit sah es aber nach etwas anderem aus...