evcc-io/evcc #23889: SHM: uppercase deviceId disconnects legacy evcc devices from SMA Portal

Martin-Becker, 2025-09-27:
### Describe the bug

Because of the changes of #23616 (#23190), legacy SEMP deviceIds are handed out uppercase to SHM, even if they are entered lowercase to new config ui for SHM integration.

So prior ID of:

`<DeviceId>F-abcAAAAa-bbbbbbbbbbbb-00</DeviceId>`

Is hand out now like:

 `<DeviceId>F-ABCAAAAA-BBBBBBBBBBBB-00</DeviceId>`

This means that the ivcc device is recognized as a new device for the SHM/SMA portal, and you will lose the connection to the previous device in the portal and thus the history.

Regardless of correctness in the sense of the SEMP protocol, it would be nice to NOT change the manually via config ui specified ID. Only newly generated IDs may be uppercase. This would ensure backward compatibility, and existing configurations could retain their history in the SMA Portal.

### Steps to reproduce

1. Configure an DeviceID in eccc 0.208.0+ SHM config ui with lower case letters
2. Look at SEMP XML on evcc semp URL: http://192.168.xxx.yyy:7070/semp/
3. Compare with SEMP XML output of evcc 0.207.6 where an lowercase ID is not uppercased
4. (ID is interpreted case-sensitive by SHM / SMA Portal)...

### Configuration details

```yaml
not necessary
```

### Log details

```text
not necessary
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

hybe2022, 2025-09-27:
Siehe auch Diskussion: 
https://github.com/evcc-io/evcc/discussions/23736#discussioncomment-14530442

andig, 2025-09-28:
@Martin-Becker could you please open a ticket with SMA to get feedback?

hybe2022, 2025-09-28:
@andig 
Das kann Monate dauern. Eigentlich ist es doch bewiesen, da es bei Personen mit lower case in der ID jetzt nicht mehr klappt...

closed 2025-09-28 by andig

andig, 2025-09-28:
Yep, da waren wir wohl zu optimistisch...

andig, 2025-09-28:
@naltatis wir könnten immer noch nur SMA-seitig lowercase sprechen und dennoch auf HEX bestehen. Das würde mir sogar noch besser gefallen. PR dafür folgt bei Gelegenheit.