evcc-io/evcc #23917: Problem with SHM 2.0 since last updates

Reaper866, 2025-09-28:
### Describe the bug

I was using EVCC via Home Assistant for the last months and also the (arguably not very good) SMA eCharger was working fine, EVCC prevented using the battery for the car (which the SHM2.0 alone isn't capable of), really great work from you guys! 

Since the last update 0.208 I got problems with getting the addon started in HA, even if I didn't make any changes before. I already tried different configurations for the hems in the evcc.yaml, but nothing worked. I removed the ", tried without any vendorid or device id. Removed the hems and/or grid part completely. Nothing worked to get it back to work :-( no programmer myself, so I hoped someone here can help me with my problem

hems:

shm:
  allowcontrol: false
  vendorid: "xxxxxxxx"
  deviceid: "xxxxxxxxxxxx"

### Steps to reproduce

1.Update to 0.208
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 60s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- name: grid
  type: template
  template: sma-home-manager 
  usage: grid  
  host: xxx 
  
- name: pv
  type: template
  template: sma-hybrid
  usage: pv
  modbus: tcpip
  id: 3
  host: xxx
  port: xxx
  
- name: battery
  type: template
  template: sma-hybrid
  usage: battery
  modbus: tcpip
  id: 3
  host: xxx
  port: xxx
  
chargers:
- type: template
  template: smaevcharger 
  host: xxx 
  user: xxx
  password: xxx
  name: wallbox

vehicles:
- type: template
  template: hyundai 
  title: Hyundai Ioniq 5  
  user: xxx
  password: xxx
  capacity: 58  
  language: de  
  name: ev

loadpoints:
- title: eCharger
  charger: wallbox
  vehicle: ev
  mode: now

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.33 # EUR/kWh
  feedin:
    type: fixed
    price: 0.08 # EUR/kWh
  co2:
    type: template
    template: grünstromindex
    zip: xxxxx
    token: xxx
  solar:
    type: template
    template: solcast
    site: Homebase
    token: xxx

site:
  title: Homebase
  meters:
    grid: grid
    pv:
    - pv
    battery:
    - battery

hems:

shm:
  allowcontrol: false
  vendorid: "xxx"
  deviceid: "xxx"
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/09/28 14:22:14 evcc 0.208.1
[main  ] INFO 2025/09/28 14:22:14 using config file: /config/evcc.yaml
[db    ] INFO 2025/09/28 14:22:14 using sqlite database: /data/evcc.db
[main  ] INFO 2025/09/28 14:22:14 UI listening at :7070
[tariff] DEBUG 2025/09/28 14:22:17 using cache: grünstromindex-xxxx (start: 2025-09-28 13:00:00 +0200 CEST, end: 2025-10-02 01:00:00 +0200 CEST)
[tariff] ERROR 2025/09/28 14:22:17 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/09/28 14:22:17 creating tariff solar-template-0 failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 429 (Too Many Requests)
[tariff] ERROR 2025/09/28 14:22:17 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/09/28 14:22:17 creating tariff solar-template-1 failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 429 (Too Many Requests)
[lp-1  ] DEBUG 2025/09/28 14:22:17 charge total import: 1647.101kWh
[main  ] FATAL 2025/09/28 14:22:17 hems failed configuring hems: breaking change: Sunny Home Manager integration is always on. See https://github.com/evcc-io/evcc/releases and https://docs.evcc.io/en/docs/integrations/sma-sunny-home-manager
[main  ] FATAL 2025/09/28 14:22:17 will attempt restart in: 15m0s

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/09/28 14:28:23 evcc 0.208.1
[main  ] INFO 2025/09/28 14:28:23 using config file: /config/evcc.yaml
[db    ] INFO 2025/09/28 14:28:23 using sqlite database: /data/evcc.db
[main  ] INFO 2025/09/28 14:28:23 UI listening at :7070
[tariff] DEBUG 2025/09/28 14:28:27 using cache: grünstromindex-xxx (start: 2025-09-28 13:00:00 +0200 CEST, end: 2025-10-02 01:00:00 +0200 CEST)
[tariff] ERROR 2025/09/28 14:28:27 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/09/28 14:28:27 creating tariff solar-template-0 failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 429 (Too Many Requests)
[tariff] ERROR 2025/09/28 14:28:27 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/09/28 14:28:27 creating tariff solar-template-1 failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 429 (Too Many Requests)
[lp-1  ] DEBUG 2025/09/28 14:28:27 charge total import: 1647.543kWh
[main  ] FATAL 2025/09/28 14:28:27 meter [grid] cannot create meter 'grid': cannot create meter type 'template': cannot create meter type 'sma': login failed: no packet received in timeout
failed booting site: not found: grid
[main  ] FATAL 2025/09/28 14:28:27 will attempt restart in: 15m0s

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/09/28 14:40:39 evcc 0.208.1
[main  ] INFO 2025/09/28 14:40:39 using config file: /config/evcc.yaml
[db    ] INFO 2025/09/28 14:40:39 using sqlite database: /data/evcc.db
[main  ] INFO 2025/09/28 14:40:39 UI listening at :7070
[tariff] DEBUG 2025/09/28 14:40:42 using cache: grünstromindex-xxxx (start: 2025-09-28 13:00:00 +0200 CEST, end: 2025-10-02 01:00:00 +0200 CEST)
[tariff] ERROR 2025/09/28 14:40:42 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/09/28 14:40:42 creating tariff solar-template-1 failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 429 (Too Many Requests)
[tariff] ERROR 2025/09/28 14:40:42 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/09/28 14:40:42 creating tariff solar-template-0 failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 429 (Too Many Requests)
[lp-1  ] DEBUG 2025/09/28 14:40:42 charge total import: 1648.420kWh
[main  ] FATAL 2025/09/28 14:40:42 hems failed configuring hems: breaking change: Sunny Home Manager integration is always on. See https://github.com/evcc-io/evcc/releases and https://docs.evcc.io/en/docs/integrations/sma-sunny-home-manager
[main  ] FATAL 2025/09/28 14:40:42 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.208

andig, 2025-09-28:
> hems failed configuring hems: breaking change: Sunny Home Manager integration is always on. See https://github.com/evcc-io/evcc/releases and https://docs.evcc.io/en/docs/integrations/sma-sunny-home-manager

Bitte die Mühe machen und die Release Notes als auch Logfile zu lesen!

closed 2025-09-28 by andig

montyburns1981, 2025-11-28:
Hey, hast du das Problem lösen können? steige bei dem gleichen Problem auch nicht durch...

andig, 2025-11-28:
Es muss einfach nur der alte `shm` eintrag aus der yaml config gelöscht werden. Fertig.

Reaper866, 2025-12-05:
```
[lp-1 ] WARN 2025/12/05 10:55:18 charger logic error: disabled but charging
[site ] DEBUG 2025/12/05 10:55:19 solar forecast: accumulated 0.015Wh from 2025-12-05 10:54:19 +0100 CET to 2025-12-05 10:55:19 +0100 CET
[site ] DEBUG 2025/12/05 10:55:19 solar forecast: produced 80.389
[site ] DEBUG 2025/12/05 10:55:19 solar forecast: accumulated 138.525kWh, produced 80.389kWh, scale 0.580
[site ] DEBUG 2025/12/05 10:56:17 ----
[lp-1 ] DEBUG 2025/12/05 10:56:17 charge power: 10854W
[circuit-main] DEBUG 2025/12/05 10:56:17 power: 10637W
[circuit-main] DEBUG 2025/12/05 10:56:17 current: 15.7A
[site ] DEBUG 2025/12/05 10:56:17 grid power: 10637W
[site ] DEBUG 2025/12/05 10:56:17 grid powers: [3505 3478 3654]W
[site ] DEBUG 2025/12/05 10:56:17 grid currents: [15 14.8 15.7]A
[site ] DEBUG 2025/12/05 10:56:17 battery 1 power: 0W
[site ] DEBUG 2025/12/05 10:56:17 pv 1 power: 538W
[site ] DEBUG 2025/12/05 10:56:17 battery 1 soc: 8%
[site ] DEBUG 2025/12/05 10:56:17 !! solar production: accumulate set pv 13465.305kWh meter total (was: Accumulated: 80.389kWh updated: 2025-12-05 10:55:18 +0100 CET meter: 13465.297kWh)
[site ] DEBUG 2025/12/05 10:56:17 !! solar production: accumulate moved pv from 80.389 to 80.397
[site ] DEBUG 2025/12/05 10:56:17 site power: 10737W
[lp-1 ] DEBUG 2025/12/05 10:56:17 session energy: 5.309kWh
[lp-1 ] DEBUG 2025/12/05 10:56:17 charger status: C
[lp-1 ] DEBUG 2025/12/05 10:56:17 soc estimated: 39.44% (vehicle: 36.00%)
[lp-1 ] DEBUG 2025/12/05 10:56:17 vehicle soc: 39%
[lp-1 ] DEBUG 2025/12/05 10:56:17 vehicle soc limit: 90%
[lp-1 ] DEBUG 2025/12/05 10:56:17 vehicle range: 112km
[lp-1 ] WARN 2025/12/05 10:56:17 charger logic error: disabled but charging 
```

[image]

Klappt damit soweit auch, erhalte jedoch immer einen charger logic error beim laden jede minute und blicke nicht ganz durch woran das liegt... aber solange es geht

andig, 2025-12-05:
@Reaper866 hier gehts um SHM, also off-topic.