evcc-io/evcc #23918: Sungrow Hausbatteriespeicher speist nach beendeter Netzladung direkt ins öffentliche Netz zurück

BennHelm, 2025-09-28:
### Describe the bug

Hallo Team, 

wie in Thread https://github.com/evcc-io/evcc/discussions/23294 eingegrenzt, bitte ich um Unterstützung und Lösung meines Problems.

Eckdaten:
- Sungrow Hybridwechselrichter SH6.0RT
- Sungrow Wallbox AC011E-01
- Sungrow Batterie: SBR128
- Kommunikationsmodul: WiNet-S
- Home Assist aus rpi5 mit EVCC als eigene Applikation
- aktuelle Firmwarestände

In EVCC besteht die Möglichkeit den Hausbatteriespeicher in Abhängigkeit vom Strompreis zu laden.
Der Beginn des Ladens und die Ausführung funktioniert prima.

Am Ende des Vorgangs oder dem vorzeitigen Beenden des Ladevorgangs (Umstellen von Preisgrenze auf "keine" im Auswahlmenü) entlädt die Hausbatterie ins öffentliche Netz, bis sie leer ist.

Nach Eingrenzung und Untersuchung von zugänglichen Konfigurationswerten in der Sungrow-Konfigurationsoberfläche zeigt sich, dass abschließende Werte nicht richtig gesetzt werden.
Es liegt gemäß Logfiles die Mutmaßung durch LeonSol nahe, dass ev. falsche Register im Modbus beschrieben werden könnten. Dies bitte ich zu prüfen und zu ändern, um die Funktion des Aufladens der Hausbatterie aus dem öffentlichen Netz nutzen zu können.

Anmerkung, das Laden des Fahrzeugs funktioniert wie gewünscht.

Ich bitte um Hilfe und Nachricht, ob weitere Informationen neben dem benannten Thread erforderlich sind, um den Fall zu lösen. 

Für Rückfragen bzw. persönliche Gespräche stehe ich gern zur Verfügung.

Vielen Dank!
MfG
Ben

### Steps to reproduce

Meine EVCC.yaml liegt anbei, sie ist reduziert um Kommentare und reduziert um persönliche Nutzerdaten.

1. Setzen des geeigneten Preisschwellwertes in EVCC UI, um Hausbatterie Ladevorgang zu starten.
2. Nach Erreichen von 100% hält Hausbatterie die 100% und bezieht für Hausverbraucher Energie aus PV oder dem Netz, so lange, bis Preisgrenze nicht mehr zutrifft oder der manuell die Preisgrenze verändert wird auf -keine-.
3. Manuelles Setzen der Preisgrenze im UI auf -keine- 
Um 2025/09/11 13:26 habe ich den Wert in der EVCC-Bedienoberfläche Preisgrenze auf -keine- geändert.

Beiliegende Trace-Loginformationen sind entstanden.
Im oben benannten Thread liegen weiterführende Loginformationen. (kpl. Doku Ein-/ Ausschalten/ Fehler/ Abhilfe).

4. Fehlersichtung, dass Endladung ins Netz stattfindet
Der Fehler ist reproduzierbar, innerhalb der Wechselrichterkonfiguration steht der EMS-Modus auf "Zwangsmodusbetrieb" und der Unterpunkt 3-2 auf 6kW. 

[image]

5. Ein manuelles Stoppen und Setzen dieses Wertes auf 0 unterbricht die Entladung ins Netz. 

[evcc-20250911-132757-trace_nach-STOPP_13h26min.log](https://github.com/user-attachments/files/22582854/evcc-20250911-132757-trace_nach-STOPP_13h26min.log)

### Configuration details

```yaml
# sponsoring token, valid from 06.07.2025 15000 Tage
sponsortoken: xxx

site:
  title: Home-Swi3
  meters:
    grid: my_grid
    pv:
      - my_pv
    battery:
      - my_battery

loadpoints:
  - title: Swi38
[... 389 lines cut ...]
[sungrow] TRACE 2025/09/11 13:27:36 modbus: send 34 d4 00 00 00 06 03 04 53 44 00 01
[sungrow] TRACE 2025/09/11 13:27:36 modbus: recv 34 d4 00 00 00 05 03 04 02 00 01
[lp-1  ] DEBUG 2025/09/11 13:27:36 charger status: A
[httpd ] TRACE 2025/09/11 13:27:41 GET /api/health
[providerauth] TRACE 2025/09/11 13:27:46 publishing 0 auth providers
[providerauth] TRACE 2025/09/11 13:27:56 publishing 0 auth providers
[httpd ] TRACE 2025/09/11 13:27:56 GET /api/system/log
[httpd ] TRACE 2025/09/11 13:27:57 GET /api/system/log
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.6

LeonSol, 2025-09-28:
Nach dem Austausch vieler Informationen mit Ben ist dies als möglich Ursache geblieben:

Post vom 11.09.2025:

@naltatis ich brauche hier einmal kurz deine Hilfe:

Nach dem erfolgreichen Netzladen der Batterie steht im log "set battery mode: normal. Danach beginnt die Batterie jedoch sich in das Netz zu entladen (der Fehler ist reproduzierbar). Ich habe jetzt in das template sungrow-hybrid.yaml geschaut und mir ist folgendes aufgefallen:

**_Beim Laden wird das Register 13051 für charge power genutzt (funktioniert im Zusammenspiel mit 13049 und 13050 auch)_**
```
source: const
value: 0xAA # Charge
set:
source: modbus
{{- include "modbus" . | indent 10 }}
register:
address: 13050 # Charge/discharge command
type: writesingle
decode: uint16
{{- if .maxchargepower }}
- source: const
value: {{ .maxchargepower }} # W
set:
source: modbus
{{- include "modbus" . | indent 10 }}
register:
address: 13051 # Charge power
type: writesingle
decode: uint16
```
**_Beim Entladen wird jedoch nicht 13051 (im Zusammenspiel mit 13049 und 13050) genutzt, sondern 33047:_**
```
source: const
value: 0 # Self-consumption mode (Default)
set:
source: modbus
{{- include "modbus" . | indent 10 }}
register:
address: 13049 # EMS mode selection
type: writesingle
decode: uint16
- source: const
value: 0xCC # Stop (Default)
set:
source: modbus
{{- include "modbus" . | indent 10 }}
register:
address: 13050 # Charge/discharge command
type: writesingle
decode: uint16
# Set max battery discharge power, effectively stops discharging
- source: const
value: 1 # 0.01kW, min allowed value for register
set:
source: modbus
{{- include "modbus" . | indent 10 }}
register:
address: 33047 # Battery max discharge power
type: writesingle
decode: uint16
```
Kann dies der Grund für die Batterieentladung in das Netz sein? In den erweiterten Parametern der Sungrow-Anlage steht der EMS-Modus dann immer noch auf Zwangsmodusbetrieb und nicht auf Eigenverbrauch.

harmendp, 2025-09-29:
Seems like there is an external EMS set?

BennHelm, 2025-09-29:
@harmendp, hi there. 
What is your idea with this: "Did you try changing the mode in HA?"? 
Where is the point I could have another try or change different values? 
I only have the values inside of EVCC in use, inside of Home Assistant I didn't make changes to the sungrow-settings until now. 
I only used the change of values inside of sungrow webinterface. Would you have a screenshot or better description for your idea? Because I do not remember a possible setting change option inside of evcc. And in Home Assit I've not tried befor or configured.
CU, Ben

LeonSol, 2025-09-29:
@harmendp do you use also Sungrow inverter and battery?

harmendp, 2025-09-29:
Yes. But I have a single phase one, so settings are a little different in some places. I also did some tests with grid charging of the home battery by setting a price, but I don't have a similar issue (except for the wrong register being reset at the end).

EVCC sets 3 registers/settings when switching to grid charging of the battery:
 - 13049 EMS mode -> Forced mode
 - 13050 Charge/Discharge -> Charge
 - 13051 charge power -> maxacpower if set, else max rating of inverter (adr 5627)

When going back to normal mode, again three registers are (re)set
 - 13049 EMS mode -> Self-consuming
 - 13050 Charge/Discharge -> Stop (default)
 - 33047 Battery max discharge power -> maxacpower if set, else 10,6kW

It is not really clear from all the logs, what values these settings have before & after the issue. But I do see some modbus errors:
[sungrow] TRACE 2025/09/11 13:26:06 modbus: close connection: write tcp 172.30.1.22:54912->172.30.1.140:502: write: broken pipe
So maybe there is some kind of conflict. Can we maybe try without the charger connected in evcc? It uses the same address, yet a different id. Maybe some conflict there? However, how it can go from forced charge to forced discharge is a mystery to me.

The same manipulations can be done from a sungrow integration in HA. It would be nice to check if they can be modified without issues from outside evcc. It can also visualize all these settings, so it would be easier to debug.

BennHelm, 2025-09-29:
Hi, until now I've no direct sungrow-integration to HA. 
I've read abaout it and I'm not sure if I want to start a next "experimental" way...
I would enter values in my start config.yaml of HA, if I understood right. 
So, if there would be an other way...

The thinking of double used adresses I think is explainable, because the car-charger is connected to the inverter via RS485. So the "Hub" for every communication is the inverter. It differentiates via the "id". And the charger works as it should do.

Unusual is the writing on different registers like written obove.

So, if there is any idea to get better information with the existing setup, I would be happy.
Otherwise am I thankful for a step by step idea, how to start a new testenvironment.

Also I could start the test again without the connected car-charger, It would be easy to write a couple of "#" in front of the lines inside of yaml.

I know, it was a lot of documentation, but did you had a look to the other discussion, I had before (Thread https://github.com/evcc-io/evcc/discussions/23294)? There the most useful information for this thread are from. May be I missed out any information, you would like to think obout? 
Thanks a lot!
Ben

premultiply, 2025-10-01:
So? [#24022](https://github.com/evcc-io/evcc/pull/24022)
Bitte testen!

BennHelm, 2025-10-03:
[2025-10-03_Doku.pdf](https://github.com/user-attachments/files/22679628/2025-10-03_Doku.pdf)
[evcc-20251003-125714-trace_12h56m30s_manueller-Befehl-in-sungrow_Eigenverbrauch.log](https://github.com/user-attachments/files/22679629/evcc-20251003-125714-trace_12h56m30s_manueller-Befehl-in-sungrow_Eigenverbrauch.log)

Leider noch kein Erfolg. 
Vorgehen anbei dokumentiert.

[image]

Danke für Eure weitere Analyse!
Weitere Überprüfungen kann ich leider erst wieder ab übernächster Woche durchführen.

Viele Grüße
Ben

harmendp, 2025-10-03:
Several modbus errors again in the log, so not all registers are being set/reset correctly.
In this case, apparently the charging power was not set to zero if I read the docu correctly.

@BennHelm What is the ip-address of the RaspPi?

LeonSol, 2025-10-03:
@premultiply Warum wird Register 33047 genutzt, wenn der Batterie-Modus auf normal gesetzt wird. Das ist doch nicht nötig. Über 13049, 13050 und 13051 ist doch alles regelbar?

harmendp, 2025-10-03:
> [@premultiply](https://github.com/premultiply) Warum wird Register 33047 genutzt, wenn der Batterie-Modus auf normal gesetzt wird. Das ist doch nicht nötig. Über 13049, 13050 und 13051 ist doch alles regelbar?

To reset values coming from 'hold' mode.
Normal mode can be called coming from 2 different modes: charging mode & hold mode.
Both use different registers.

BennHelm, 2025-10-03:
looks better without capacity...

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/10/03 23:09:15 evcc 0.208.1 (278027875)
[main  ] INFO 2025/10/03 23:09:15 using config file: /config/evcc.yaml
[db    ] INFO 2025/10/03 23:09:15 using sqlite database: /data/evcc.db
[main  ] INFO 2025/10/03 23:09:16 UI listening at :7070
[site  ] INFO 2025/10/03 23:09:17 site config:
[site  ] INFO 2025/10/03 23:09:17   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/10/03 23:09:17     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/10/03 23:09:17     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/10/03 23:09:17     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/10/03 23:09:17   vehicles:
[site  ] INFO 2025/10/03 23:09:17     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/10/03 23:09:17   tariffs:
[site  ] INFO 2025/10/03 23:09:17     grid:      priceforecast
[site  ] INFO 2025/10/03 23:09:17     feed-in:   pricestatic
[site  ] INFO 2025/10/03 23:09:17     co2:       ✗
[site  ] INFO 2025/10/03 23:09:17     solar:     ✓
[lp-1  ] INFO 2025/10/03 23:09:17 loadpoint 1:
[lp-1  ] INFO 2025/10/03 23:09:17   mode:        off
[lp-1  ] INFO 2025/10/03 23:09:17   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/10/03 23:09:17   meters:      charge ✓
[lp-1  ] INFO 2025/10/03 23:09:17     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2025/10/03 23:09:17 vehicle updated: unknown -> Bens-ID7
[site  ] WARN 2025/10/03 23:09:17 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] INFO 2025/10/03 23:09:18 car disconnected

BennHelm, 2025-10-03:
Yaml inklusive „max charge power“

23:11 -> Ladestart

23:13:10 -> Umstellung auf -keine-

Result is the same.

[evcc-20251003-231350-trace_entlaedt-weiter.log](https://github.com/user-attachments/files/22688216/evcc-20251003-231350-trace_entlaedt-weiter.log)

harmendp, 2025-10-03:
Hello,

I found some similar issues, see link below. Can you check the status on your battery while discharging to the grid in isolarcloud. Did you notice jumps in SOC-level?

[https://www.photovoltaikforum.com/thread/173108-sungrow-sbr096-entl%C3%A4dt-ins-netz-einspeisung/](url)

According to these posts, could be normal behaviour for the battery to recalibrate, discharge from 100 to empty. Would explain a lot.

LeonSol, 2025-10-19:
@premultiply geht es hier noch weiter oder ist keine Lösung in Sicht?

BennHelm, 2025-10-21:
Hello @harmendp, I've read the post too.
I didn't find a way to let the charge process do it by itsself. 
So the Information is totaly good to understand, but my sungrow app shows nowhere the icon, that there is a maintenance mode (As I understood, the only hint I could get to realize this procedure.)

If the battery gets empty after ending the chargingprocess it only goes down to 0%, there is no automaticly restart in charging (!).

So I would think, that I did not interrupt any mechanism. 

As I understood now, the current firmware has a little change in its behaviour and uses the "normal" limits (100% max and 10% min) to recalibrate automaticyaly.

BennHelm, 2025-10-21:
[image]

I tried again.
If the lowest value is riched nothing else happens. Everything with the battery is "blocked", there is no recharging with sunpower or anything else...

I hope this answer helps to exclude the recalibrating option...

BennHelm, 2025-10-21:
Hello @premultiply you asked for the "right" order of commands...

After stop of charging sungrow web UI looks like this:

[image]

Now I have to switch manually to "Erzwungener Modus", than there are new sub-settings:

[image]

I switch them manually to "Stopp" and 0kW.

[image]

After this is done I switch again to "Eigenverbrauch".

harmendp, 2025-10-21:
If need be, I can try to create a pull request to change the order of the commands. Maybe we need to change the ems mode last in stead of first.

BennHelm, 2025-10-28:
Hello @harmendp , 
I did the manuall steps and documentated them in the attached files.

Sorry, but they are in german.
I hope the pictures give the necessary information.

I only are able to do this inside of sungrow.
I hope, you will have the right answer about the registers/ adresses inside of modbus...

Is there more I could do?

[2025-10-28_manuelles-Laden.pdf](https://github.com/user-attachments/files/23190790/2025-10-28_manuelles-Laden.pdf)
[Parameterabfrage_A2291366939_20251028125338_Ausgangszustand.pdf](https://github.com/user-attachments/files/23190793/Parameterabfrage_A2291366939_20251028125338_Ausgangszustand.pdf)
[Parameterabfrage_A2291366939_20251028153915_100Prozent-Ladeendzustand.pdf](https://github.com/user-attachments/files/23190792/Parameterabfrage_A2291366939_20251028153915_100Prozent-Ladeendzustand.pdf)
[Parameterabfrage_A2291366939_20251028154410_nach-Stopp-Befehl.pdf](https://github.com/user-attachments/files/23190791/Parameterabfrage_A2291366939_20251028154410_nach-Stopp-Befehl.pdf)
[Parameterabfrage_A2291366939_20251028154909_Ruhezustand-nach-Ladung.pdf](https://github.com/user-attachments/files/23190789/Parameterabfrage_A2291366939_20251028154909_Ruhezustand-nach-Ladung.pdf)

harmendp, 2025-10-31:
There was a fix done yesterday for modbus communication with Sungrow.
Maybe not a bad idea to test again once there is a new release.

BennHelm, 2025-10-31:
Hi @harmendp, good idea, I'll try again and tell you the result :-)

BennHelm, 2025-11-01:
Hi @harmendp, I just testet the actual behaviour. No change.... After I change the value for money to "keine", the energy goes directly to the grid. So, depending on your new test and havong a look at the right order of commands I'm curious about new ideas...

Here are the log-details:

[2025-11-01_Test-with-actual-nightly-version.txt](https://github.com/user-attachments/files/23282574/2025-11-01_Test-with-actual-nightly-version.txt)

harmendp, 2025-11-03:
> Hi [@harmendp](https://github.com/harmendp), I just testet the actual behaviour. No change.... After I change the value for money to "keine", the energy goes directly to the grid. So, depending on [...]

Would you mind to try another test? 

Start battery charge from grid via evcc, and then in iSolarcloud set ems-mode to selfConsumption. This is the first change evcc would do when going back to normal mode. Would be interesting to know if it also starts to feed-in to the grid.

If so, a second test can be done. First start battery charge from grid via evcc, then change in iSolarCloud
 - force charge/discharge to stop
 - ems-mode to selfConsumption

Thanks!

BennHelm, 2025-11-06:
Hi @harmendp, 
I did your 2 tests.
1. Starting inside of EVCC -> inside of Sungrow switch to "Selfconsumption" -> same behaviour -> the battery gives energy direct to the grid. 

1.b) direct try to "overwrite" with new setting insiede of evcc with a higher value of price, no change, it continues giving its own energy to the grid. 
1.c) I need to set to "keine" inside of EVCC, than after it I could start the next try

2. test:
a) set a value inside of evcc, battery loads out of the grid. 
b) inside of sungrow: manual set of "force discharge - stop" -> it stopped
c)  inside of sungrow: manual set of "selfConsumption" -> everything works as it should

Sounds like the second command-order is the right one!

Have a good day!

harmendp, 2025-11-22:
Hello @BennHelm The PR was merged yesterday. So it would be nice if you could try the last nightly or the next release and check if there is any improvement. Please keep us posted.

harmendp, 2025-11-23:
This is very strange. I'm running out of ideas to be honest. We are now doing exactly the same as your manual test in iSolarCloud (test 2). There are no modbus errors in the logs, so all writes are being accepted.

Can you once again (sorry!) send us the parameters of the inverter just before & after the charging of the battery from the grid and the discharging of the battery to the grid. Also, I also still think it would be a good idea to test without the charger attached. It might interfere in the modbus comm between evcc & the inverter.

Thanks!

harmendp, 2025-11-25:
I don't quite understand the summary graph from iSC, this one looks OK, no?:

[image]
[image]

We don't need to reset the charge/discharge power, as the mode controls if it is used or not. Also, there is no discharge mode in EVCC anywhere. See my settings:

[image]

Changing an extra setting just for a single user is maybe not the best idea. There is another test we can do, to be sure this setting is really the cause. 6kW happens to be the max the battery can do. So can you set maxchargepower to a lower value (2500W f.e.) in evcc config. This is the value that is written to the max charge/discharge power in forced mode. Does the battery then after returning to normal mode discharge with 2.5kW to the grid or does it still do 6kW?

Thanks!