evcc-io/evcc #23941: Solax x3 Hybrid G4 Battery hold not working in fast an plan mode

farcorben, 2025-09-29:
### Describe the bug

Hallo liebes Team,

mir ist aufgefallen das im Moment der "schnell" und plan lade Modus (nach Preis) aktuell nicht die Batterie sperren kann um das entladen zu verhindern. Der Fehler ist folgender:

`battery mode: modbus: response unit id '0' does not match request '1'`

### Steps to reproduce

1. schalter: verhindern der batterieentladung aktivieren

[image]

2. Lademodus "schnell" oder einen Plan erstellen nach Preis
3. Fehler kommt instand nach ladestart und batterie wird nicht gesperrt.
...

### Configuration details

```yaml
29|2|template|{"baudrate":19200,"defaultmode":"0","host":"192.168.1.11","id":"1","maxacpower":"0","modbus":"tcpip","mppt3":"false","storageunit":1,"template":"solax","usage":"grid"}|||Solax Hybrid X1/X3 G3/G4
37|2|template|{"baudrate":19200,"capacity":12,"defaultmode":"5","host":"192.168.1.11","id":"1","maxacpower":"0","modbus":"tcpip","mppt3":"false","storageunit":1,"template":"solax","usage":"battery"}|Solax_t30||Solax Hybrid X1/X3 G3/G4
38|2|template|{"baudrate":19200,"defaultmode":"0","host":"192.168.1.11","id":"1","maxacpower":6600,"modbus":"tcpip","mppt3":"false","storageunit":1,"template":"solax","usage":"pv"}|Solax_pv||Solax Hybrid X1/X3 G3/G4
```

### Log details

```text
main ] INFO 2025/09/29 12:00:39 evcc 0.208.1 (f9adcfa3)
[main ] INFO 2025/09/29 12:00:39 using config file: /etc/evcc.yaml
[db ] INFO 2025/09/29 12:00:39 using sqlite database: /root/.evcc/evcc.db
[mqtt ] INFO 2025/09/29 12:00:39 connecting evcc-1223032740 at tcp://192.168.0.254:1883
[mqtt ] DEBUG 2025/09/29 12:00:39 tcp://192.168.0.254:1883 connected
[main ] INFO 2025/09/29 12:00:39 UI listening at :7070
[lp-1 ] DEBUG 2025/09/29 12:00:39 set title: Solax EVC
[lp-1 ] DEBUG 2025/09/29 12:00:40 set priority: 0
[lp-1 ] DEBUG 2025/09/29 12:00:40 set smart cost limit: <nil>
[lp-1 ] DEBUG 2025/09/29 12:00:40 set smart feed-in limit: <nil>
[lp-1 ] DEBUG 2025/09/29 12:00:40 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:500}}
[lp-1 ] DEBUG 2025/09/29 12:00:40 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1 ] DEBUG 2025/09/29 12:00:40 set session energy limit: 0
[lp-1 ] DEBUG 2025/09/29 12:00:40 set session soc limit: 0
[lp-1 ] DEBUG 2025/09/29 12:00:40 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc000c9bedc}
[lp-1 ] DEBUG 2025/09/29 12:00:40 set default mode: pv
[lp-1 ] DEBUG 2025/09/29 12:00:40 set phases: 3
[lp-1 ] DEBUG 2025/09/29 12:00:40 set min current: 6
[lp-1 ] DEBUG 2025/09/29 12:00:40 set max current: 16
[tariff] DEBUG 2025/09/29 12:00:40 using cache: solcast-99c880787fbcd995450e1208bd72dae2c6ea09de15b6358824cc0d1c1e089f8d (start: 2025-09-29 12:00:00 +0200 CEST, end: 2025-10-01 13:00:00 +0200 CEST)
[tariff] DEBUG 2025/09/29 12:00:40 using cache: grünstromindex-e05b92857cf6cf624bd98b18de210ce78670b52132d390724bb82fcf916a64af (start: 2025-09-29 11:00:00 +0200 CEST, end: 2025-10-02 01:00:00 +0200 CEST)
[lp-1 ] DEBUG 2025/09/29 12:00:40 charge total import: 10014.500kWh
[site ] WARN 2025/09/29 12:00:41 vehicle 'LeapMotor T03' does not support automatic detection
[site ] INFO 2025/09/29 12:00:41 site config:
[site ] INFO 2025/09/29 12:00:41 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/09/29 12:00:41 grid: power ✓ energy ✓ currents ✗
[site ] INFO 2025/09/29 12:00:41 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2025/09/29 12:00:41 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/09/29 12:00:41 vehicles:
[site ] INFO 2025/09/29 12:00:41 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site ] INFO 2025/09/29 12:00:41 vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/09/29 12:00:41 tariffs:
[site ] INFO 2025/09/29 12:00:41 grid: priceforecast
[site ] INFO 2025/09/29 12:00:41 feed-in: ✗
[site ] INFO 2025/09/29 12:00:41 co2: ✓
[site ] INFO 2025/09/29 12:00:41 solar: ✓
[lp-1 ] INFO 2025/09/29 12:00:41 loadpoint 1:
[lp-1 ] INFO 2025/09/29 12:00:41 mode: pv
[lp-1 ] INFO 2025/09/29 12:00:41 charger: power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/09/29 12:00:41 meters: charge ✓
[lp-1 ] INFO 2025/09/29 12:00:41 charge: power ✓ energy ✓ currents ✓
[... 82 lines cut ...]
[mqtt ] TRACE 2025/09/29 12:01:09 send evcc/site/batteryGridChargeActive: 'false'
[mqtt ] TRACE 2025/09/29 12:01:09 send evcc/loadpoints/1/chargeCurrents: '23.8'
[mqtt ] TRACE 2025/09/29 12:01:09 send evcc/loadpoints/1/chargeCurrents/l1: '8.02'
[mqtt ] TRACE 2025/09/29 12:01:09 send evcc/loadpoints/1/chargeCurrents/l3: '7.95'
[mqtt ] TRACE 2025/09/29 12:01:09 send evcc/loadpoints/1/chargeCurrents/l2: '7.83'
[db:37 ] TRACE 2025/09/29 12:01:09 modbus: send 00 0f 00 00 00 06 01 06 00 20 00 00
[db:37 ] TRACE 2025/09/29 12:01:10 modbus: close connection: modbus: response unit id '0' does not match request '1'
[site ] ERROR 2025/09/29 12:01:10 battery mode: modbus: response unit id '0' does not match request '1'
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.208.1 (f9adcfa3) - nightly.20250928-f9adcfa

andig, 2025-09-29:
Da stellt sich die Frage: wie wäre es denn richtig (laut eigenem Experiment oder Doku)?

farcorben, 2025-09-29:
@andig , ich glaube es geht wieder ... 
Ich habe den WR "rekonfiguriert" und die Modbus IDs in WR "überschrieben" ... 

Jetzt geht es wieder ... Vielleicht ein fehlerhaftes update? Wer weiß... aber sicher kein evcc Thema.

danke