evcc-io/evcc #23945: IDM Wärmepumpe - Konsumierte Energiemenge bzw. Kosten werde nicht mehr protokolliert / berechnet

james2810, 2025-09-29:
### Describe the bug

Seit 24.09. (letzter Eintrag unter Ladevorgänge, Zusammenhang mit Update auf 0.208??) wird bei meiner Wärmepumpe die konsumierte Energiemenge bzw. die damit verbundenen Kosten nicht mehr protokolliert/berechnet.

Letzter Eintrag:
[image]

Aktuelle Anzeige:
[image]

Woran kann dies liegen??

### Steps to reproduce

Keine Schritte notwendig.
Sobald die Wärmepumpe läuft, tritt das Problem auf.

### Configuration details

```yaml
Charger: 
  - name: idm_charger
    type: template
    template: idm
    modbus: tcpip
    id: 1
    host: 192.168.xxx.xxx # Hostname
    port: 502 # Port
    tempsource: warmwater_bottom # optional    

Loadpoint:
- title: IDM # display name for UI
    charger: idm_charger # charger
    circuit: main
```

### Log details

```text
idm_charger:

[idm_charger] TRACE 2025/09/29 14:07:44 modbus: send 05 b0 00 00 00 06 01 03 10 1a 00 02
[idm_charger] TRACE 2025/09/29 14:07:44 modbus: recv 05 b0 00 00 00 07 01 03 04 31 27 3f f8
[idm_charger] TRACE 2025/09/29 14:07:49 modbus: send 05 b9 00 00 00 06 01 03 10 20 00 02
[idm_charger] TRACE 2025/09/29 14:07:49 modbus: recv 05 b9 00 00 00 07 01 03 04 34 c9 46 af
[idm_charger] TRACE 2025/09/29 14:07:49 modbus: send 05 bb 00 00 00 06 01 03 10 20 00 02
[idm_charger] TRACE 2025/09/29 14:07:49 modbus: recv 05 bb 00 00 00 07 01 03 04 34 c9 46 af
[idm_charger] TRACE 2025/09/29 14:07:49 modbus: send 05 bd 00 00 00 06 01 03 10 20 00 02
[idm_charger] TRACE 2025/09/29 14:07:49 modbus: recv 05 bd 00 00 00 07 01 03 04 34 c9 46 af
[idm_charger] TRACE 2025/09/29 14:07:49 modbus: send 05 bf 00 00 00 06 01 03 03 f4 00 02
[idm_charger] TRACE 2025/09/29 14:07:49 modbus: recv 05 bf 00 00 00 07 01 03 04 6c 49 42 49
[idm_charger] TRACE 2025/09/29 14:07:59 modbus: send 05 d9 00 00 00 06 01 03 10 1a 00 02
[idm_charger] TRACE 2025/09/29 14:07:59 modbus: recv 05 d9 00 00 00 07 01 03 04 e9 79 3f f6
[idm_charger] TRACE 2025/09/29 14:08:14 modbus: send 06 0b 00 00 00 06 01 03 10 1a 00 02
[idm_charger] TRACE 2025/09/29 14:08:14 modbus: recv 06 0b 00 00 00 07 01 03 04 31 27 3f f8
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.208.1

james2810, 2025-09-29:
Nach welchem Bereich muss ich Ausschau halten?? Ladepunkt?? Charger??
Kann das Logfile persistent gespeichert werden?? In evcc ist immer nur ein kurzer Abriss drin.

andig, 2025-09-29:
„Ich auch“ ist auch kein Logfile

dieterlind, 2025-09-30:
Hier ein Auszug aus dem aktuellen Log:

```
[lp-3 ] DEBUG 2025/09/30 08:38:59 charge power: 514W
[lp-3 ] DEBUG 2025/09/30 08:39:14 charge power: 504W
[lp-3 ] DEBUG 2025/09/30 08:39:29 charge power: 505W
[lp-3 ] DEBUG 2025/09/30 08:39:33 charge total import: 22462.102kWh
[lp-3 ] DEBUG 2025/09/30 08:39:33 charger status: B
[lp-3 ] DEBUG 2025/09/30 08:39:33 charge final energy: 0.045kWh
[lp-3 ] DEBUG 2025/09/30 08:39:33 charge total import: 22462.102kWh
[lp-3 ] DEBUG 2025/09/30 08:39:34 pv charge current: 0.543A = 0.732A + -0.188A (130W @ 3p)
[lp-3 ] DEBUG 2025/09/30 08:39:44 charge power: 494W
[lp-3 ] DEBUG 2025/09/30 08:39:59 charge power: 492W
[lp-3 ] DEBUG 2025/09/30 08:40:14 charge power: 483W
[lp-3 ] DEBUG 2025/09/30 08:40:18 charge total import: 22462.146kWh
[lp-3 ] DEBUG 2025/09/30 08:40:18 charger status: B
[lp-3 ] DEBUG 2025/09/30 08:40:18 charge final energy: 0.045kWh
[lp-3 ] DEBUG 2025/09/30 08:40:18 charge total import: 22462.146kWh
[lp-3 ] DEBUG 2025/09/30 08:40:18 pv charge current: 0.559A = 0.7A + -0.141A (97W @ 3p)
[lp-3 ] DEBUG 2025/09/30 08:40:29 charge power: 483W
[lp-3 ] DEBUG 2025/09/30 08:40:44 charge power: 483W
[lp-3 ] DEBUG 2025/09/30 08:40:59 charge power: 483W
[lp-3 ] DEBUG 2025/09/30 08:41:03 charge total import: 22462.191kWh
[lp-3 ] DEBUG 2025/09/30 08:41:03 charger status: B
[lp-3 ] DEBUG 2025/09/30 08:41:03 charge final energy: 0.045kWh
[lp-3 ] DEBUG 2025/09/30 08:41:03 charge total import: 22462.191kWh
[lp-3 ] DEBUG 2025/09/30 08:41:03 pv charge current: 0.555A = 0.7A + -0.145A (100W @ 3p)
[lp-3 ] DEBUG 2025/09/30 08:41:14 charge power: 483W
[lp-3 ] DEBUG 2025/09/30 08:41:29 charge power: 482W
[lp-3 ] DEBUG 2025/09/30 08:41:44 charge power: 483W
[lp-3 ] DEBUG 2025/09/30 08:42:01 charge total import: 22462.252kWh
[lp-3 ] DEBUG 2025/09/30 08:42:01 charger status: B
[lp-3 ] DEBUG 2025/09/30 08:42:01 charge final energy: 0.061kWh
[lp-3 ] DEBUG 2025/09/30 08:42:02 charge total import: 22462.252kWh
[lp-3 ] DEBUG 2025/09/30 08:42:02 pv charge current: 0.555A = 0.7A + -0.145A (100W @ 3p)
[lp-3 ] DEBUG 2025/09/30 08:42:02 charge power: 492W
[lp-3 ] DEBUG 2025/09/30 08:42:14 charge power: 534W
[lp-3 ] DEBUG 2025/09/30 08:42:29 charge power: 565W
[lp-3 ] DEBUG 2025/09/30 08:42:32 charge total import: 22462.283kWh
[lp-3 ] DEBUG 2025/09/30 08:42:32 charger status: B
[lp-3 ] DEBUG 2025/09/30 08:42:33 charge final energy: 0.031kWh
[lp-3 ] DEBUG 2025/09/30 08:42:33 charge total import: 22462.283kWh
[lp-3 ] DEBUG 2025/09/30 08:42:33 pv charge current: 0.659A = 0.819A + -0.159A (110W @ 3p)
[lp-3 ] DEBUG 2025/09/30 08:42:44 charge power: 594W
[lp-3 ] DEBUG 2025/09/30 08:42:59 charge power: 634W
[lp-3 ] DEBUG 2025/09/30 08:43:14 charge power: 675W
[lp-3 ] DEBUG 2025/09/30 08:43:18 charge total import: 22462.338kWh
[lp-3 ] DEBUG 2025/09/30 08:43:18 charger status: B
[lp-3 ] DEBUG 2025/09/30 08:43:18 charge final energy: 0.055kWh
[lp-3 ] DEBUG 2025/09/30 08:43:18 charge total import: 22462.338kWh
[lp-3 ] DEBUG 2025/09/30 08:43:18 pv charge current: 0.807A = 0.978A + -0.171A (118W @ 3p)
[lp-3 ] DEBUG 2025/09/30 08:43:29 charge power: 714W
[lp-3 ] DEBUG 2025/09/30 08:43:44 charge power: 754W
[lp-3 ] DEBUG 2025/09/30 08:43:59 charge power: 794W
[lp-3 ] DEBUG 2025/09/30 08:44:05 charge total import: 22462.396kWh
[lp-3 ] DEBUG 2025/09/30 08:44:05 charger status: B
[lp-3 ] DEBUG 2025/09/30 08:44:05 charge final energy: 0.059kWh
[lp-3 ] DEBUG 2025/09/30 08:44:05 charge total import: 22462.396kWh
[lp-3 ] DEBUG 2025/09/30 08:44:05 pv charge current: 1A = 1.15A + -0.151A (104W @ 3p) 
```

andig, 2025-09-30:
Bin nicht sicher was uns das sagen soll. Insbesondere kann ich keinen Ladevorgaang erkennen.

dieterlind, 2025-09-30:
Hätte verstanden, dass die WP läuft (charge power), dass eigentlich der aktuelle Verbrauch für den Vorgang mitprotokolliert wird (charge final energy) und auch der Gesamtenergie angepasst wird (total import).

Tombra1889, 2025-09-30:
Gestern wurden wieder Verbräuche protokolliert, es tritt also nicht jeden Tag auf.

dieterlind, 2025-09-30:
Nach eine wenig herumlesen und beobachten: Kann es damit zusammenhängen, dass der Charger Status immer "B" bleibt?? 

[image]

B würde ja eigentlich bedeuten, dass die WP "nur" angeschlossen ist
C allerdings, dass geladen wird.
Siehe auch: https://github.com/evcc-io/evcc/discussions/7286

Ich habe jetzt mal versucht die Informationen für lp-3 (=Wärmepumpe) für den letzten Ladevorgang (=Warmwasseraufbereitung) aus dem Log zu extrahieren:

```
[lp-3  ] DEBUG 2025/09/29 13:00:17 charge power: 0W
[lp-3  ] DEBUG 2025/09/29 13:00:32 charge power: 74W
[lp-3  ] DEBUG 2025/09/29 13:00:40 charge total import: 22418.605kWh
[lp-3  ] DEBUG 2025/09/29 13:00:40 charger status: B
[lp-3  ] DEBUG 2025/09/29 13:00:40 charge final energy: 0.010kWh
[lp-3  ] DEBUG 2025/09/29 13:00:40 charge total import: 22418.605kWh
[lp-3  ] DEBUG 2025/09/29 13:00:40 vehicle soc: 30%
[lp-3  ] DEBUG 2025/09/29 13:00:40 pv charge current: 0A = 0.107A + -6.6A (4555W @ 3p)
[lp-3  ] DEBUG 2025/09/29 13:00:47 charge power: 506W

...

[lp-3  ] DEBUG 2025/09/29 13:01:02 charge power: 897W
[lp-3  ] DEBUG 2025/09/29 13:01:17 charge power: 1008W
[lp-3  ] DEBUG 2025/09/29 13:01:27 charge total import: 22418.658kWh
[lp-3  ] DEBUG 2025/09/29 13:01:27 charger status: B
[lp-3  ] DEBUG 2025/09/29 13:01:27 charge final energy: 0.053kWh
[lp-3  ] DEBUG 2025/09/29 13:01:27 charge total import: 22418.658kWh
[lp-3  ] DEBUG 2025/09/29 13:01:27 vehicle soc: 30%
[lp-3  ] DEBUG 2025/09/29 13:01:27 pv charge current: 0A = 1.46A + -8.05A (5555W @ 3p)
[lp-3  ] DEBUG 2025/09/29 13:01:32 charge power: 1128W

...

[lp-3  ] DEBUG 2025/09/29 14:27:59 charge power: 1768W
[lp-3  ] DEBUG 2025/09/29 14:28:07 charge total import: 22428.164kWh
[lp-3  ] DEBUG 2025/09/29 14:28:07 charger status: B
[lp-3  ] DEBUG 2025/09/29 14:28:07 charge final energy: 0.066kWh
[lp-3  ] DEBUG 2025/09/29 14:28:08 charge total import: 22428.164kWh
[lp-3  ] DEBUG 2025/09/29 14:28:08 pv charge current: 2.34A = 2.56A + -0.223A (154W @ 3p)
[lp-3  ] DEBUG 2025/09/29 14:28:14 charge power: 1768W
[lp-3  ] DEBUG 2025/09/29 14:28:29 charge power: 1767W
[lp-3  ] DEBUG 2025/09/29 14:28:44 charge power: 0W
[lp-3  ] DEBUG 2025/09/29 14:28:50 charge total import: 22428.203kWh
[lp-3  ] DEBUG 2025/09/29 14:28:50 charger status: B
[lp-3  ] DEBUG 2025/09/29 14:28:51 charge final energy: 0.039kWh
[lp-3  ] DEBUG 2025/09/29 14:28:51 charge total import: 22428.203kWh
[lp-3  ] DEBUG 2025/09/29 14:28:51 limitSoc reached: 55.1% > 55%
[lp-3  ] DEBUG 2025/09/29 14:28:59 charge power: 0W
```

Vielleicht hilft das ja weiter.

andig, 2025-10-01:
Es gibt kein Problem. Ohne Boost aufgrund PV Überschuss wird auch nichts geloggt.