evcc-io/evcc #23952: BMW - new cloud access restrictions

jpara, 2025-09-29:
### Describe the bug

Hi,

I just received a notification within the myBMW app (attached).
It seems like, this is in effect already now:

[image]
[image]

### Steps to reproduce

Configure a BMW vehicle

### Configuration details

```yaml
BMW
```

### Log details

```text
no log entries
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.208.1

andig, 2025-09-29:
It has _not_ been working for a long time or really only with horrible hcaptcha "process". We should be able to find a somewhat-working solution using the BMW EU data act streaming interface, but that will take some time to implement.

SolarPower2024, 2025-09-29:
> 
> There is also a first home assistant integration using the non-streaming API https://github.com/JjyKsi/bmw-cardata-ha
> 
> Maybe it helps as an intermediate solution.

Just tried this out and at least some sensors are showing up. 👍🏻

Satsang081, 2025-09-30:
> > > Ich kann auf CardDataClient erstellen klicken, dann muss man ja die zwei Regler auf der rechten Seite aktivieren, diese gehen nach ein paar Sekunden wieder aus und oben auf der Seite erscheint ein rotes Feld mit einem Error, man soll sich mit einem Code an die Kundenbetreuung wenden, hat es schon jemand geschafft und kann mir erklären wie?
> > 
> > 
> > Hatte keinerlei Probleme mit unserem i4. Mit " zwei Regler auf der rechten Seite" meinst du das, was man in https://bmw-cardata.bmwgroup.com/customer/public/api-documentation/Id-Technical-registration_Step-2 sieht?
> 
> Also die Checkboxes/Schalter für "Request access to CarData API" und "CarData Stream" - ja, da kommt bei mir auch ein Fehler.

Da bin ich ja froh nicht der einzige zu sein. Vorübergehend habe ich gerade meinen BMW bei evcc über Tronity eingebunden, wusste nicht das es diese option gibt, nun warten wir mal bis evcc weiter kommt und dann schauen wir mal

mfuchs1984, 2025-09-30:
> Hab das mit zwei unabhängigen Accounts probiert

Es muss meines Wissens zwingend der Hauptaccount verwendet werden.

Worog5, 2025-09-30:
> > Hab das mit zwei unabhängigen Accounts probiert
> 
> Es muss meines Wissens zwingend der Hauptaccount verwendet werden.

Ja das ist korrekt. Zumindest laut der API Beschreibung:

[image]

blenni, 2025-10-01:
Bei meinem Ladevorgangs heute Nacht wurde im Stream auch der SOC aktualisiert. Ich vermute BMW hat das behoben.

tipp88, 2025-10-01:
Bei mir funzt der MQTT Stream auch super. Ich habe im EVCC das Fahrzeug auf iobroker.bmw umgestellt. EVCC holt dann die Daten über SimpleAPI vom Iobroker BMW adapter. Der BMW Adapter ist zwar tot, aber ich aktualisiere dessen Datenpunkte vom MQTT Stream. Läuft gut. Der Stream liefert auch annähernd Realtime Daten.
Einzig blöde ist ja nur, der User Code ist nur 2 Wochen gültig, danach muss man wieder über das Portal Berechtigen. Wird es da im EVCC eine automatische Lösung geben?

SEFNSRL, 2025-10-01:
@tipp88 

kannst Du dazu vielleicht etwas im Iobroker Forum [https://forum.iobroker.net/topic/46720/test-adapter-bmw-mini-v2-0-0](url) schreiben? ich bin mir sicher Du hilfst damit einigen Usern.

Gruss S.

Kentrath, 2025-10-01:
> Ich hatte die gleichen Probelme mit der Fehlermeldung. Der "Trick", mit dem ich es geschafft habe, war...warten. Client erstellen. Eine Minute warten. Unteren Schalter. Eine Minute warten. Oberen Schalter.
> 
> War das der Trick? Keine Ahnung. Muss man eine Minute warten? Keine Ahnung. Hat es so bei mir funktioniert? Ja!

Kann mir bitte jemand helfen. Ich habe auf Car Data Client geklickt. Es baut sich ein Fenster mit einer id und den beiden Feldern auf. Ich kopiere die id und aktiviere beide Felder. Danach klicke ich auf Authentifizierung. Es öffnet sich ein neues Fenster. Hier gebe ich die gerade erstellte id ein. Es wird jedoch gemeckert, dass ich etwas eingeben soll, das im Auto Display erscheint. Da passiert aber nichts. Wie habt Ihr das gemacht?
Vielen Dank für Hilfe.

grommelmerje, 2025-10-01:
Nein, Du musst einen Code eintragen, nicht die ID. Ich konnte mir den Code hier unter Angabe meiner VIN. erstellen. Auf meinem Fahrzeugdisplay kam auch nichts: https://bmw-cardata.bmwgroup.com/customer/public/api-specification

sepv22, 2025-10-01:
Den Code kann man über den auf der BMW-Seite beschriebenen CURL erhalten:
 `curl --request POST \
  --url https://customer.bmwgroup.com/gcdm/oauth/device/code \
  --header 'Accept: application/json' \
  --header 'Content-Type: application/x-www-form-urlencoded' \
  --data client_id=cj5b3499-4918-40x6-a232-f4112f837d72 \
  --data response_type=device_code \      
  --data 'scope=authenticate_user openid cardata:streaming:read cardata:api:read'
  --data code_challenge=yourCodeChallenge \
  --data code_challenge_method=S256 \
` 

Data-Client-ID entsprechend austauschen und über das Terminal absenden.
Den HA-Adapter bekomme ich aber auch nicht zum Laufen.  Obwohl die BMW Seite eine erfolgreiche Anmeldung rückmeldet.

> Authorization is not complete yet (error: Token polling failed (403): {'error_description': 'The user has declined authorization', 'error': 'access_denied'}). Please approve the device and try again.

tipp88, 2025-10-01:
> Die verlinkte Seite ist ja alles andere als intuitiv. In welcher Rubrik hast Du denn was eingegeben? Es gibt ja Device Code Flow und CarData...

Was wollt ihr damit? Manuell nützt euch der Code doch ohnehin nichts, wenn ihr es nicht alle 60min refreshed ist der Usercode sowieso wieder weg.

michiproep, 2025-10-01:
Nur als kleine Anmerkung. Der iobroker.bmw Adapter wurde aktualisiert. v4.0.x Damit bekommt man seine Fahrzeugdaten wieder. Allerdings passt die Struktur nicht mehr zum evcc vehicle template.
Als temporären Workaround kann man im Objektbauem des iobroker adapter ein Dummy-Fahrzeug anlegen und per Skript die Datenpunken syncen. In evcc muss man dann noch die VIN ändern.

[image]

mfuchs1984, 2025-10-02:
Ich habe noch weiter mit meinem BMW i4 und den Entitäten experimentiert mit dem Home Assistant Addon https://github.com/JjyKsi/bmw-cardata-ha

**status**
`vehicle.powertrain.tractionBattery.charging.port.anyPosition.isPlugged` und `vehicle.drivetrain.electricEngine.charging.status` (`NOCHARGING, INITIALIZATION, CHARGINGACTIVE, CHARGINGPAUSED, CHARGINGENDED, CHARGINGERROR`) reagieren sofort beim Anstecken, Abstecken, Ladestart und Ladestop.

**soc**
`vehicle.drivetrain.batteryManagement.header` liefert diesen bspw. beim Ladestart und Ladestop, ansonsten eher sehr unregelmäßig, ggf. bei Überschreitung von Schwellwerten oder anderen triggern.
`vehicle.drivetrain.electricEngine.charging.level` wird nicht über MQTT streaming ausgeliefert (ist bei mir auch gar nicht im Katalog zur Auswahl), es benötigt API polling und liefert dann einen recht guten guten Wert. 

**range**
`vehicle.drivetrain.electricEngine.kombiRemainingElectricRange` liefert recht zeitnahe Range updates.
`vehicle.drivetrain.electricEngine.remainingElectricRange` ist zwar im MQTT Katalog verfügbar, verhält sich jedoch soweit ich sehe entsprechend wie `vehicle.drivetrain.electricEngine.charging.level` und braucht ein aktives API polling

**odometer:**
`vehicle.vehicle.travelledDistance` liefert diesen zuverlässig

**climater**
`vehicle.vehicle.preConditioning.activity` wechselt sofort beim ein- oder ausschalten zwischen `INACTIVE`, `HEATING `und wahrscheinlich wenns wärmer ist, `COOLING`

**limitsoc**
`vehicle.powertrain.electric.battery.stateOfCharge.target` ist auch im MQTT Katalog verfügbar, aber wird nicht regelmäßig über MQTT aktualisiert. Brauchtauch API polling.

Ich habe mir daraus ein rudimentäres custom vehicle gebaut:
<details>
```
vehicles:
  - name: i4
    type: custom
    title: "BMW i4"
    capacity: 80.7
    soc:
      source: http
      uri: http://192.168.178.77:8123/api/states/sensor.drivetrain_batterymanagement_header
      jq: .state | tonumber
      auth:
        type: bearer
        token: xxxx
    status:
      source: http
      uri: http://homeassistant.local:8123/api/states/sensor.drivetrain_electricengine_charging_status
      jq: if .state == "CHARGINGACTIVE" then "C" else "B" end
      auth:
        type: bearer
        token: xxxx
    range:
      source: http
      uri: http://homeassistant.local:8123/api/states/sensor.drivetrain_electricengine_kombiremainingelectricrange
      jq: .state
      auth:
        type: bearer
        token: xxxx
    odometer:
      source: http
      uri: http://homeassistant.local:8123/api/states/sensor.travelleddistance
      jq: .state  
      auth:
        type: bearer
        token: xxxx
    climater:
      source: http
      uri: http://homeassistant.local:8123/api/states/sensor.preconditioning_activity
      jq: if .state == "HEATING" or .state == "heating" or .state == "COOLING" or .state == "cooling" then true else false end
      auth:
        type: bearer
        token: xxxx
    limitsoc:
      source: http
      uri: http://homeassistant.local:8123/api/states/sensor.target_state_of_charge
      jq: .state
      auth:
        type: bearer
        token: xxxx
```
</details>

Ak11rA, 2025-10-02:
> Den HA-Adapter bekomme ich aber auch nicht zum Laufen. Obwohl die BMW Seite eine erfolgreiche Anmeldung rückmeldet.

Ich auch nicht. Ich habe eine Client ID auf der BMW Seite angelegt, dann mir curl den POST gesendet:

curl -X POST https://customer.bmwgroup.com/gcdm/oauth/device/code \
  -d "client_id=XXX" \
  -d "response_type=device_code" -d "scope=authenticate_user"

Und eine Antwort bekommen, aus der ich user_code und device_code extrahiert habe. In der Response-URL habe ich dann den Device Code eingetragen, und die BMW Webseite zeigt mir einen gültigen Stream:

[image]

Dabei stehen dann noch ein Topic und ein Benutzername.

Aber in HA kann ich keine Integration BimmerData Streamline einrichten, der Setup-Assistent fragt mich nach einer Client-ID, und egal, was ich da eigene, die Client ID, user_code, device_code oder den Benutzernamen des Streams, HA sagt:  "device_code_failed".

Was habe Ihr denn da eingetragen?

mfuchs1984, 2025-10-02:
In /homeassistant/configuration.yaml

Danach wahrscheinlich Neustart.

[image]

cdyna43, 2025-10-02:
Habe ich gemacht. Bringt leider nichts. 

bleibt bei "Forbidden"

Muss der Token in Anführungszeichen? Ich habe den jetzt einfach hinter "token" kopiert also anstelle der xxxx im Beispiel.

michiproep, 2025-10-04:
> > Nur als kleine Anmerkung. Der iobroker.bmw Adapter wurde aktualisiert. v4.0.x Damit bekommt man seine Fahrzeugdaten wieder. Allerdings passt die Struktur nicht mehr zum evcc vehicle template. Als temporären Workaround kann man im Objektbauem des iobroker adapter ein Dummy-Fahrzeug anlegen und per Skript die Datenpunken syncen. In evcc muss man dann noch die VIN ändern.
> > <img alt="Image" width="370" height="392" src="https://private-user-images.githubusercontent.com/12296115/496229815-757444b2-3a63-48d8-ad6e-5592fef3c642.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTk1MDkzNjQsIm5iZiI6MTc1OTUwOTA2NCwicGF0aCI6Ii8xMjI5NjExNS80OTYyMjk4MTUtNzU3NDQ0YjItM2E2My00OGQ4LWFkNmUtNTU5MmZlZjNjNjQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEwMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMDAzVDE2MzEwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmMDk5M2MyMDdjMGRiOGYzMzJlNDFmMTJmZDZhZTFhZDNjNTczNmM2MWIxOThkYWRkNWU3MWM5ZTVkYmZmZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nEJcuHp6re0K1SiM5mynSjmZ7MG2jg9cs3phXul0ieM">
> 
> Ich bekomme leider nur aus "range" und "currentMileage" Daten ausgelesen. Hast du mir bitte noch nähere Infos, welches Format o.ä.? Vielen Dank.
@Blau-Mann 
Den Ladestand und die Reichweite bekommst du aus
bmw.0.<VIN>.stream.vehicle.drivetrain.electricEngine.kombiRemainingElectricRange.value
bmw.0.<VIN>.stream.vehicle.drivetrain.batteryManagement.header.value

Format ist ne gute Frage. evcc funktioniert zwar, meldet aber warnings im log. Schau ich mir beim nächsten Laden nochmal an...

marcelGoerentz, 2025-10-04:
Sieht aus als würde BMW nun OAuth2 nutzen, hier müssten wir den Registrierungs-Flow implementieren. Gibt es jemanden der dafür prädestiniert ist? Ansonsten würde ich das machen, habe allerdings nicht wirklich Zeit dafür...

hangingloose, 2025-10-04:
> Die neue integration steht doch schon zum Test bereit, siehe [#23972](https://github.com/evcc-io/evcc/pull/23972)

@andig 
kannst du abschätzen, wie lange es dauern wird, bis das im Update drin ist... 
Oder gibt es irgendwo eine DAU Anleitung für die neue Integration?

tilibibi, 2025-10-05:
Ich habe das gleiche Problem mit meinem BMW & Mini ... 
Habe gerade versucht, den CarData Client zu erstellen - was man zwingend braucht, wenn ich das richtig verstanden habe. 
Ich bekomme im BMW Portal aber immer die Meldung: "Es ist ein Fehler aufgetreten. Bitte versuchen Sie es später erneut oder wenden Sie sich an die BMW Kundenbetreuung."
Hattet ihr das auch? Oder ist das System gerade überlastet, weil alle gleichzeitig wollen?!

mfuchs1984, 2025-10-05:
@tilibibi du musst es probieren bis es geht, und dem System zwischen deinen Klicks Zeit lassen, das läuft noch nicht stabil. Details siehe https://github.com/evcc-io/evcc/discussions/24117#discussioncomment-14596097

tilibibi, 2025-10-05:
danke ... ich versuche weiter ... 
bisher noch kein Erfolg :-(
Danke euch allen aber für den schnellen Fix!

PTiTDaV, 2025-10-05:
Hi all,

I have an issue with the Cardata implementation. I have 2 BMW's linked to 2 different accounts.
I followed the guide for both Cars on each account but when i try to authorize the second car i get an error "Failed Login : unidentified"
I the logs i also get errors.
```
[providerauth] DEBUG 2025/10/05 12:48:22 login request for: BMW/Mini CarData (xxxxxxxx)
[oauth-xxxxxxx] TRACE 2025/10/05 12:48:22 POST https://customer.bmwgroup.com/gcdm/oauth/device/code
[oauth-xxxxxxx] TRACE 2025/10/05 12:48:23 client_id=xxxxxxxxxxxxxxxx&code_challenge=xxxxxxxxxxxxxxxxxxxx&code_challenge_method=S256&scope=authenticate_user+openid+cardata%3Astreaming%3Aread+cardata%3Aapi%3Aread -- {"error": "invalid_request", "error_description": "The request is missing a required parameter, includes an unsupported parameter value (other than grant type), repeats a parameter, includes multiple credentials, utilizes more than one mechanism for authenticating the client, or is otherwise malformed"}
```

Delta1977, 2025-10-05:
Ich bekomme keine PIN im Auto angezeigt.

marcelGoerentz, 2025-10-05:
> Ich bekomme keine PIN im Auto angezeigt. 

Das funktioniert ohne PIN im Auto.
Der Flow ist wie folgt:
- Im BMW connected drive zur Fahrzeugübersicht gehen und beim Fahrzeug dann auf CarData gehen. Dort einen neuen Client erzeugen.
Setzt die Schalter nacheinander. Bei mir war ein Button ausgegraut bevor ich den nächsten Schalter umgelegt habe, habe ich gewartet bis der Button wieder Blau wurde
- Kopiert die Client ID und fügt sie in evcc bei eurer Konfiguration ein.
- Startet den Autorisierungsprozess über den Menüpunkt CarData sollte bei euch noch gelb sein:
[image]
Ihr werdet auf eine externe Seite weitergeleitet und der Autorisierungsprozess läuft im Hintergrund automatisch
- Am Ende sollte dann in etwa stehen, dass der Client erfolgreich registriert wurde. Und der Gelbe Punkt ist dann Grün, wie bei meinem Screenshot.

andig, 2025-10-05:
Es läuft noch nicht ganz rund, weitere Verbesserung folgt.

hangingloose, 2025-10-05:
Trotzdem von mir und sicher vielen anderen hier ein dickes DANKE für die schnelle Hilfe!

jeloneal, 2025-10-05:
Tausend Dank für eure Mühen, jetzt hab auch ich verstanden wies geht 👍

andig, 2025-10-05:
Ruhe bewahren, evcc aktualisieren, neu starten.

PTiTDaV, 2025-10-05:
Is it possible i'm missing this datapoint ?
vehicle.drivetrain.electricEngine.charging.level