evcc-io/evcc #23984: Please add Wallbox EVECUBE 2S Wallbox| 2x22 kW

DevineCZ, 2025-09-30:
Hi, I would like to add support for the EVECUBE 2S wallbox to evcc. I have the OpenAPI specification. The wallbox supports modbus, OCPP and HTTP API. The modbus and OCPP support is not complete, but over http it has all the features I need. However, I would prefer if there was native support, not just over http. I would be happy to help with testing or provide the information you need to add wallbox support to EVCC.
Thanks in advance for your help.

[openapi-evecube.yaml](https://github.com/user-attachments/files/22626618/openapi-evecube.yaml)

DevineCZ, 2025-09-30:
There is an integration into HA available here, if OpenAPI is not enough, some things can be found here:

https://github.com/TeepCo/chuck_control/tree/develop/custom_components/chuck_control

andig, 2025-10-01:
Why not use the existing OCPP integration? Does "not complete" mean that it doesn't work?

premultiply, 2025-10-01:
@DevineCZ Please try #24019

DevineCZ, 2025-10-02:
Hello, here is the output from each endpoint. The car is not connected and is not charging. So some values ​​may be zero.

/api/status

[
  {
    "id": 1,
    "status": "Unavailable",
    "voltage": 0,
    "voltages": [0, 0, 0],
    "current": 0,
    "maxCurrent": 10,
    "energy": 0,
    "energyTotal": 0.025,
    "carConnected": false,
    "phasesCurrent": [0, 0, 0]
  },
  {
    "id": 2,
    "status": "Unavailable",
    "voltage": 0,
    "voltages": [0, 0, 0],
    "current": 0,
    "maxCurrent": 6,
    "energy": 0,
    "energyTotal": 1931.204,
    "carConnected": false,
    "phasesCurrent": [0, 0, 0]
  }
]

/api/admin/status

{
  "connectors": [
    {
      "id": 1,
      "status": "Unavailable",
      "voltage": 0,
      "voltages": [0, 0, 0],
      "current": 0,
      "currents": [0, 0, 0],
      "energy": 0,
      "energyTotal": 0.025,
      "carConnected": false,
      "lastStatusPacket": {
        "connectorId": 1,
        "evseId": 230570,
        "carStatus": "NOT_CONNECTED",
        "chargingStatus": "NOT_CHARGING",
        "voltage": 243.8,
        "voltages": [244.2, 243, 244.2],
        "current": 0,
        "actualWh": 0,
        "totalWh": 25,
        "lockStatus": "UNKNOWN",
        "internalTemperature": 23,
        "externalTemperature": 0,
        "warnings": [
          "EXTERNAL_CURRENT_MEASUREMENT_COMMUNICATION_ERROR"
        ],
        "errors": [
          "ALL_OK"
        ],
        "extraInfo": {
          "external_meter_current_l1": 0,
          "external_meter_current_l2": 0,
          "external_meter_current_l3": 0,
          "current_l1": "0.0",
          "current_l2": "0.0",
[... 213 lines cut ...]

/api/info

{
  "vendor": "EV Expert",
  "model": "EVECUBE 2S Wallbox| 2x22 kW",
  "version": "2.4.0",
  "revision": "b46e29ae16",
  "serialNumber": "202310700027",
  "isManualTransactionStartSupported": false,
  "connectors": [
    {
      "id": 1,
      "type": "SERIAL",
      "connection": "",
      "sendResetOnStart": true
    },
    {
      "id": 2,
      "type": "SERIAL",
      "connection": "",
      "sendResetOnStart": true
    }
  ]
}